Marus/cortex-debug #374: Flash software breakpoints are not cleared when exiting debug session

Preebluda, 2021-01-15:
While debugging my code for Cortex-M0+ MCU, I found out that "Cortex-Debug" doesn't clear flash software breakpoints when it closes debug session. That results in corrupted flash, so after power reset MCU breaks in the place of software breakpoint even when you are not debugging.

I haven't found any previous mentions of this problem, so I decided to report.

Preebluda, 2021-01-15:
I suppose this issue will be solved with pull request "delete all break points with disconnect." #363 by @kohada88

Marus, 2021-01-17:
There is a 0.3.8-pre release on the GitHub release page that you can install to get what should hopefully fix this issue immediately. I hope to get a final 0.3.8 release out in the next day or two once I finish testing a few things.

Marus, 2021-01-26:
@Preebluda - I'm going to assume that this has fixed your issues with the flash corruption and close the issue. If you're still having issues please re-open the ticket and we can investigate further.

closed 2021-01-26 by Marus