Marus/cortex-debug #377: Unable to debug with FreeRTOS awareness enabled

bryce-cahill, 2021-01-18:
Hi there,

Up until recently I was able to debug fine using the following config:

```
                "name": "X Debug Release",
                "cwd": "${workspaceFolder:X}",
                "executable": "${workspaceFolder:X}/project/X/build/release/X.elf",
                "request": "launch",
                "type": "cortex-debug",
                "servertype": "jlink",
                "device": "STM32L451VE",
                "runToMain": false,
                "rtos": "FreeRTOS",
                "svdFile": "${workspaceFolder:X}/toolchain/STM32L4x1.svd"
```

However, recently when trying to launch this config I get the following log: `"monitor" command not supported by this target.` Removing the line `"rtos": "FreeRTOS",` allows debugging to launch as usual but without FreeRTOS awareness.

arm-none-eabi-gdb version: 8.3.0.20190709-git
JlinkGDBServer version: V6.94a

This was working well up until recently and I'm unsure what has changed. Any help is much appreciated, thanks.

Marus, 2021-01-19:
Hum, this is an interesting one - haven't been using FreeRTOS on anything recently - so haven't used that functionality much - but don't know of anything that would be likely to have caused a change to the handling of that launch.json configuration recently. Have you updated your J-Link GDB server recently? It's possible that there is some change in more recent versions of the J-Link server that is causing issues.

I'll try to get some time to try this with the newer J-Link GDB servers (as mine is kind of out-of-date at the moment).

Also, can you try adding `"showDevDebugOutput": true` to your launch.json configuration and posting the output (which will likely be quite verbose - may want to post it as an attachment) - might give a bit more detail on exactly what's happening.

haneefdm, 2021-01-19:
I haven't used JLink with an RTOS. You have to specify a command-line option to specify a plugin.

https://www.segger.com/products/debug-probes/j-link/tools/j-link-gdb-server/thread-aware-debugging/
https://forum.segger.com/index.php/Thread/5595-SOLVED-SEGGER-FreeRTOS-plugin-with-FreeRTOS-v10-0-0-no-thread-found/

So, the `rtos` option has to be a full pathname to the dll/so plugin. Not just the name of the RTOS. Please try that and let us know

Marus, 2021-01-19:
@haneefdm - The plugin normally should be mapping the RTOS name and loading the appropriate plugin based upon the configuration property.

bryce-cahill, 2021-01-19:
@Marus it could be due to JLinkGDBServer updating but unfortunately I wouldn't be able to tell you which version it was last working on.

Here is the verbose log, I have intentionally replaced some of the file paths with "X" https://gist.github.com/bryce-cahill/8972c9ddbdb444e0987f55dacca9b394

haneefdm, 2021-01-19:
@Marus Not sure how to map rtos name to a specific plugin dll/so. Given there are other rtoses that don't even come with a JLink release.

bryce-cahill, 2021-01-21:
As an update this I noticed that when I try and debug when FreeRTOS awareness I get the following logs in the Adapter Output
```
...
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 2 bytes @ address 0x080498A4 (Data = 0x807D)
Read 4 bytes @ address 0x0801E68C (Data = 0xBF00E7E5)
Reading register (s0 = 0xBE7918CE)
*** buffer overflow detected ***: terminated
```

Marus, 2021-01-26:
@bryce-cahill - one more question for you, what version of FreeRTOS are you using in your project?

Marus, 2021-01-26:
> @Marus Not sure how to map rtos name to a specific plugin dll/so. Given there are other rtoses that don't even come with a JLink release.

Yes, I was just stating that for the two RTOS that j-link supports out of the box (namely FreeRTOS and Seggers own embOS) we map the two names allowed in the launch.json configuration to the appropriate DLL when launching -  namely line 170 of configprovider.ts (apparently they are now bundling a plug-in for ChibiOS as well now so in theory I could enable that support for J-Link).  At the moment we don’t support third-party RTOS plug-ins for J-Link (they do have an SDK for doing them, although I’ve never looked for other ones being implemented as I’ve pretty much only used FreeRTOS myself)

bryce-cahill, 2021-01-27:
> @bryce-cahill - one more question for you, what version of FreeRTOS are you using in your project?

@Marus based off Amazon FreeRTOS release v1.3.2

Marus, 2021-01-27:
I just tried one of my old FreeRTOS test beds that I just use for debugging cortex-debug itself (uses V8.2.1 of FreeRTOS - I really should update the test bed mind you - but don't have time for that right at the moment) with the very latest J-Link GDB server for macOS (v6.94b) and at least that test bed app seems to debug fine with me getting proper stacks for all the tasks in my test bed. 

SEGGER's documentation states that they support FreeRTOS versions v8 and above, I'm not really sure how the amazon FreeRTOS versions match up with the "Classic" FreeRTOS versions (they both seemed to have moved to <YEAR><MONTH>.<RELEASE> version numbers recently). Based upon the dates of the release on GitHub for amazon-freertos (https://github.com/aws/amazon-freertos/releases) and "standard" FreeRTOS (https://github.com/FreeRTOS/FreeRTOS/releases?after=PartialReleases) - amazon v1.3.2 and "standard" v10.1.0 both seem to have been released on August 22, 2018 - which makes me think they are effectively the same release (with whatever specific changes the amazon one includes).

The reason I bring that specific version up is this support article - https://www.freertos.org/FreeRTOS_Support_Forum_Archive/August_2018/freertos_v10.1.0_breaks_thread_aware_debugging_with_JLink_d406aae6j.html - in which V10.1.0 breaks thread aware debugging with J-Link

@bryce-cahill - If you have time - I'd suggest attempting to change your freertos version - it's possible that it is just some compatibility issue with that version on the J-Link server.

Marus, 2021-01-27:
FYI - that version broke compatibility with some other debug servers as well including OpenOCD

bryce-cahill, 2021-01-27:
Interesting find @Marus. However, I just manually downgraded the jlink-software-and-documentation package from version v6.94 to v6.88a (a version that I found I had working in the past) and thread aware debugging is working again. Unfortunately without going through all the versions in-between I would be hard pressed to tell you which version of the package broke things.

Thanks for having a look at this.

closed 2021-01-27 by bryce-cahill

Marus, 2021-01-27:
Interesting - as the v6.94 doesn't break it in general for all FreeRTOS versions - but certainly seems to have issues with that particular one. On our side, we can't do anything about it as it just seems to crash the GDB server - which at that point puts it out of our hands. Glad you found a work-around.