Marus/cortex-debug #379: Using openocd and stlink-dap.cfg bundled with stm32cubeide, stm32h750 cannot be debugged normally.

yueduz, 2021-01-19:
But it is possible in stm32cubeide, and swo can also be used in stm32cubeide

Marus, 2021-01-19:
Don't really have much information to go off of here - as I don't know the particular variant of openOCD (and there are unfortunately a lot of OpenOCD variants) or have the cfg file that is being used. 

Can you provide some more details including the cfg file and your launch.json configuration.

yueduz, 2021-01-19:
This is a program to light up the LED, the program does not stop at the main() function, nor does it light up the LED.

[image]
-------------------------------------------------
launch.json
 ```
{
    // 使用 IntelliSense 了解相关属性。 
    // 悬停以查看现有属性的描述。
    // 欲了解更多信息，请访问: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "OpenOCD_STLink",
            "type": "cortex-debug",
            "executable": "${workspaceRoot}/build/yueyi.elf",
            // "cwd": "/usr/share/openocd/scripts",
            // "cwd": "${workspaceRoot}",
            "cwd": "/opt/stm32cubeide/plugins/com.st.stm32cube.ide.mcu.debug.openocd_1.5.0.202011091203/resources/openocd/st_scripts",
            "request": "launch",
            "servertype": "openocd",
            "svdFile": "${workspaceRoot}/STM32H750x.svd",
            "preLaunchTask": "make", //先运行Build任务
            "runToMain": true,
            "configFiles": [
                "${workspaceRoot}/yueqi Debug.cfg",
            ],
            // "swoConfig":{
            //     "enabled":true,
            //     "source":"probe",
            //     "swoFrequency": 2000000,
            //     "cpuFrequency":480000000,
            //     "decoders": [
            //         {
            //             "port": 0,
            //             "type": "console",
            //             "label": "SWO output",
            //             "encoding":"ascii"
            //         }
            //     ]
            // }
        },
    ]
}
```

--------------------------------------------------------
yueqi Debug.cfg
```
# This is an genericBoard board with a single STM32H750VBTx chip
#
# Generated by STM32CubeIDE
# Take care that such file, as generated, may be overridden without any early notice. Please have a look to debug launch configuration setup(s)

source [find interface/stlink-dap.cfg]

set WORKAREASIZE 0x8000

transport select "dapdirect_swd"

set CHIPNAME STM32H750VBTx
set BOARDNAME genericBoard

# Enable debug when in low power modes
set ENABLE_LOW_POWER 1

# Stop Watchdog counters when halt
set STOP_WATCHDOG 1

# STlink Debug clock frequency
[... 682 lines cut ...]

proc stm32h7x_cti_start_cm4_cti {} {
	global _CHIPNAME
	global $_CHIPNAME.CM4_CAN_HALT
	global $_CHIPNAME.CM4_CAN_BE_HALTED

	set can_halt [expr [set $_CHIPNAME.CM4_CAN_HALT] ? 0x4 : 0]
	set can_be_halted [expr [set $_CHIPNAME.CM4_CAN_BE_HALTED] ? 0x1 : 0]

	stm32h7x_cti_config_one cm4_cti on $can_halt $can_be_halted
}

proc stm32h7x_cti_stop_one {cti} {
	stm32h7x_cti_config_one $cti off 0 0
}

proc stm32h7x_cti_prepare_restart_one {cti} {
	global _CHIPNAME

	# Acknowlodge EDBGRQ at TRIGOUT0
	$_CHIPNAME.$cti write INACK 0x01
	$_CHIPNAME.$cti write INACK 0x00
}

```

closed 2021-05-02 by yueduz