Marus/cortex-debug #382: Escaped double quotes in postLaunchCommands = 'Problem parsing arguments' error

japaric, 2021-01-19:
The upcoming OpenOCD 0.11.0 supports a new set of rtt commands. One of the commands requires that its last argument is wrapped in double quotes, e.g. `rtt setup 0x20000000 49152 "SEGGER RTT"`.

Specifying that command with escaped double quotes in the `postLaunchCommands` results in a parsing error.

This is the `launch.json` file I used:

``` json
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./target/thumbv7em-none-eabihf/debug/stm-app",
            "name": "Debug Microcontroller (launch)",
            "request": "launch",
            "preLaunchTask": "rust: cargo build",
            "type": "cortex-debug",
            "runToMain": true,
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32l4x.cfg"
            ],
            "servertype": "openocd",
            "postLaunchCommands": [
                "monitor rtt setup 0x20000000 49152 \"SEGGER RTT\"", // <-
                "monitor rtt start",
                "monitor rtt server start 8765 0",
            ],
            "svdFile": "STM32L4x3.svd"
        }
    ]
}
```

I get the following pop-up error message when I try to use the above launch configuration:

``` text
Failed to launch GDB: Problem parsing arguments: interpreter-exec console "monitor rtt setup 0x20000000 49152 "SEGGER RTT"" (from interpreter-exec console "monitor rtt setup 0x20000000 49152 "SEGGER RTT"")
```

Is there a way to execute OpenOCD commands that contain double quotes *after* the `main` breakpoint has been hit (i.e. with `"runToMain": true` in the launch.json)?

### Workaround

As a workaround I'm currently using an external GDB script file.

``` json
            "postLaunchCommands": [
                "source rtt.gdb",
            ],
```

``` console
$ cat rtt.gdb
monitor rtt setup 0x20000000 49152 "SEGGER RTT"
monitor rtt start
monitor rtt server start 8765 0
```

Marus, 2021-01-19:
When we send this we have to wrap it in an appropriate GDB MI command - (interpreter exec "...") - and looks like we aren't doing an escape of the input there. Should be an easy fix.

I don't know - you could also try using single quotes as well - possible GDB my interpret that correctly - but will work on fixing the escape (as the escaping you're entering as at the JSON level).

You could also try as a work around entering it as `"monitor rtt setup 0x20000000 49152 \\\"SEGGER RTT\\\"" - that may work as well

Marus, 2021-01-19:
Note that if that second one does work now - once I put in the appropriate escaping you'll end up needing to remove that double escape - but could be a slightly cleaner workaround until such a fix is released.

Marus, 2021-01-20:
@japaric - sorry - had some encoding issues there with my comment so it didn’t come out right - fixed now - but in essence was to make an extra encoded slash - but like I said - not sure that will work for sure. Will look into the encoding issue though to get a permanent fix.

japaric, 2021-01-20:
thanks for the prompt reply!

wrapping `SEGGER RTT` in single quotes doesn't work. It is accepted by the extension and transmitted by GDB but then rejected by OpenOCD due to wrong syntax.

adding escaped backslashes around the escaped double quotes works perfectly though!

> Will look into the encoding issue though to get a permanent fix.

once that's fixed will `"monitor rtt setup 0x20000000 49152 \\\"SEGGER RTT\\\""` continue working? or will it need to be changed to use only the escaped quotes `"monitor rtt setup 0x20000000 49152 \"SEGGER RTT\""`? (either way it's not much of an issue to me; I'll add a comment in my `launch.json` to remind myself that the escaped backslashes may need to be removed in the future)

Marus, 2021-01-20:
Yeah - once I put in a proper encoding fix (not sure if I expect this will come in 0.3.9 - as I've already started the 0.3.8 preparations then you'll have to remove that extra `\\` - but just thought that might be a nicer work-around than the source variation until I can do that.

Marus, 2021-01-22:
@japaric  - If you try the 0.3.9 version just released this should fix the escapes so that you don't have to do the double escapes in `postLaunchCommands` etc (don't try the 0.3.8 - in an ill-advised attempt to include this fix in 0.3.8 I broke postLaunchCommands completely )

closed 2021-01-22 by Marus

japaric, 2021-01-25:
thanks for the quick fix, 0.3.9 works nicely!