Marus/cortex-debug #383: Enable SWO Data from Segger J-Trace

coherent-lyn, 2021-01-20:
I wish to enable the J-Trace SWO Listener port so that I can process raw SWO data when debugging.
How can I send a "monitor SWO EnableTarget 0 0 1 0 command to the J-Trace when the debugger is launched?

Marus, 2021-01-20:
You should be able to use the `postLaunchCommands` to send these additional commands to the GDB session.

Should be as simple as adding

```
"postLaunchCommands": [
  "monitor SWO EnableTarget 0 0 1 0"
],
```

to the launch.json configuration (replace `postLaunchCommands` with `postAttachCommands` if you're using "request": "attach")

coherent-lyn, 2021-01-20:
Thank you, that seems work correctly.
That was what I tried to do, but I made a mistake in the syntax.

Marus, 2021-01-20:
Ok - I'll close this issue. Thanks for letting me know it worked.

closed 2021-01-20 by Marus