Marus/cortex-debug #386: Enhancement:

manuargue, 2021-01-22:
It would be nice to display peripheral information parsed from the SVD file that is currently not visible. This allow to have a quick reference without going to the reference manual. I propose to have an optional view below the current peripheral view. Below is the information it could show when selecting each of the different fields:

Peripheral:
```
Peripheral:            RNG
Base address:          0x50060800

Description:
Random number generator
```

Peripheral/Register:
```
Register:              CR
Address:               0x50060800
Value:                 0x0
Size:                  32
Reset value:           0x0
Reset mask:            0xFFFFFFFF
Access permission:     RW
Read action:		

Description:
control register
```

Peripheral/Register/Field:
```
Bit field:         IE
LSB:               3
MSB:               4
Size:              1
Reset value:       0x0
Access permission: RW
Read action:		

Description:
Interrupt enable
```

If you think it adds value I can definitely work on it.

Marus, 2021-01-22:
My intention is to add this kind of detail to the tool tips on the tree view. This will require me to finish the upgrade in the vscode dependency as I think that is only available on newer versions of the vscode runtime.

manuargue, 2021-01-23:
Yes it makes sense having this details in the tool tips since probably is not something that we would look so often.

manuargue, 2021-01-23:
Also the register could show in the tool tips the value in different formats. Sometimes is useful to see value in an specific format quickly. For example:
``
Name : r0
Hex:0xc0000000
Decimal:-1073741824
Octal:030000000000
Binary:11000000000000000000000000000000
```

RisinT96, 2021-01-23:
VisualGDB have a great feature where when you click on the register, if the SVD has information about the purpose/names of the bits of the register, it will show you a table, with the bits mapped to their names and actual values.

Marus, 2021-01-25:
@manuargue - I do like the idea about the different formats being presented in the tooltip, I think that will be great addition. 

And yes, I do like the idea of giving information about the fields of individual registers in the tooltip summary for the register. 

Hopefully can start work on this throughout this week - and hopefully have something (even if it isn’t a final version) to release in the near future.

Marus, 2021-01-28:
@manuargue and @RisinT96 - If you want you could install and try out the V0.3.11-pre1 version from the GitHub releases page. This has the start of the enhancements to the peripheral register viewer - with a lot more detail in the tooltips for register and field level nodes (currently haven't added any detail to the top-level peripheral nodes yet - as that requires a little bit more work to do as in those cases it doesn't always have the information loaded - so need to do some additional steps to load data and a few other things if I enhance those).

Anyway - if you get a chance do have a look and let me know what you think of these enhancements and if you have ideas for further improvements.

manuargue, 2021-01-28:
I just gave it a try, it looks awesome! My only observation is that for the registers and field level, it would be nice to show also the reset value and access permission when available.

manuargue, 2021-01-28:
And do you think this can be extended to the registers view as well? Mostly to show the register value in different formats at once.

Marus, 2021-01-28:
Reset value and access level should be doable - do think those are good additions.

Probably should be extendable to the core registers view as well, although might do that in the next release - will see if I have time to do that addition right away.

Marus, 2021-01-29:
@manuargue - So I've added the reset value - on a second line below the name, address, etc.

For access level what I've done is the following: 

- If it is read/write - I don't display anything special
- If it is write only - it display's <Write Only> like the summary in the tree itself does in place of the value
- If it is read only - between the Name/Address and the current value it displays a (Read Only) label

I'll likely do another pre-release tomorrow sometime, and then assuming I don't find any issues in my testing do a final release this coming weekend sometime.

manuargue, 2021-01-29:
That's great! let me know if I can help with some testing on this.

Marus, 2021-01-30:
Hey @manuargue  - Unfortunately wasn't feeling well so didn't get to doing this yesterday as I had planned - but a second pre-release has been uploaded to to the GitHub page now - you can feel free to give it another run-through. Unless issues are found will likely release tomorrow eventing or Monday.

manuargue, 2021-01-31:
No worries :) I did some manual testing on tag `v0.3.11-pre2` and everything worked as expected. Thanks addressing this feature request in such a short time, it looks great by the way!

Marus, 2021-01-31:
Sounds good. Planning on doing the final release of v0.3.11 later today.

And @manuargue - thanks for all your PRs and helpful replies to issues.

closed 2021-02-01 by Marus