Marus/cortex-debug #388: STM32F4 + libopencm3 odd/undefined behaviour

dowensagain, 2021-01-27:
Good evening!

First, thank you for this wonderful project.

My issue is simple, but I've reached the end of my troubleshooting rope so I'm hoping someone can tell me where I'm going wrong.

### My Setup
- ST STM32F407G Discovery board with on-board ST-Link V2.1 debugger
- Visual Studio Code 1.52.1
- Latest version of Cortex-Debug
- st-util
- OpenOCD
- arm-none-eabi v10

### The Problem
Debugging an example from libopencm3-examples results in odd behavior such as:
1. Ignored breakpoints
2. `undefined` stop reasons: "Not implemented stop reason (assuming exception): undefined"
3. Halting consistently at line 67 of `vector.c` of `libopencm3`
3. Line stepping intermittently working, pausing not working

### What I've Tried
- I've tried several examples such as `fancyblink` and `miniblink`, as well as my own files. The result is the same.
- I've tried Texan's st-util as well as OpenOCD, both result in identical behavior
- I have verified that the programs I am attempting to debug build, flash, and operate correctly on the target chip
- I have loaded the .elf files into GDB via st-util and OpenOCD directly and did not encounter the halting behavior
- I tried removing `-03` from the make options; no change in behavior
- Verified the device and st-link have valid entries as reported by `lsusb`
- Provided st-util with a .svd file for my chip
- Enabling plugin debugging output, but I wasn't able to glean useful information from the product

### Steps to reproduce
1. Clone https://github.com/libopencm3/libopencm3-examples 
2. `make`
3. Set a breakpoint in `fancyblink.c`
4. Attempt to debug `libopencm3-examples/examples/stm32/f4/stm32f4-discovery/fancyblink/fancyblink.elf`

Below is my`launch.json` for reference. I'm rather new to the embedded world so I'm hoping I'm missing something painfully obvious. 

Thank you!

`{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ST-Util",
            "cwd": "${workspaceRoot}",
            "executable": "./fancyblink.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "stutil",
            "svdFile": "/home/daniel/STM32F4_svd_V1.2/STM32F407.svd"
        },
        {
            "name": "OpenOCD",
            "cwd": "${workspaceRoot}",
            "executable": "./fancyblink.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": ["interface/stlink-v2-1.cfg", "target/stm32f4x.cfg"],
        }
    ]
}`

Marus, 2021-02-05:
I don't have either STM32F4 board, but just tried the the `fancyblink` example for one of the F103 boards that I do have. 

I actually tried debugging this outside of the extension just with the command line tools, using both a J-Link debugger and ST-Link V2+OpenOCD - and in both cases the examples do not run properly; end up with the PC going to 0xDEADBEEE. (I had removed the `-Os` from the build arguments in an effort to get it to function - but no help).

So, I don't think the issue lies in the extension - but something with the examples and/or how they are being built. I unfortunately am not really familiar with libopencm3 more than the level of "I knew it existed" - not one I've ever looked at or used.

Have you tried a basic debug session using GDB on the command-line - I'm guessing it won't work either - which would indicate like I said - that it isn't an issue with the extension - but something at a lower level the examples.

Wish I had more to suggest to you; but unfortunately I don't have time at the moment to dig through the libopencm3 examples and build to see why it might be that they aren't debugging properly at the GDB layer.

dowensagain, 2021-02-21:
Hey Marus,

Thanks so much for your reply and the time you spent investigating this. Since my original post I've been able to move past my issues and I now have a working build + debug process. It boiled down to two separate issues, neither of which had to do with Cortex-Debug:

1. The flags I was passing to `arm-none-eabi-gcc` did not include `-g -ggdb3`. With these flags all debug information is included in the binary as well as a full list of sources, which also allowed me to debug an assembly *.s which exposes a function in the C file.
2. The halting on `vector.c` issue still occurs, but I have minimized the inconvenience by adding `"until -source ${fileBasename} -line 1"` to the `postStartSessionCommands` field in `launch.json` which skips the halt automatically.

I was unable to get this working using st-link, but OpenOCD works very well so far. In the process of troubleshooting my issue I was forced to learn a lot about GDB and the gcc-based toolchain, so I am not too disappointed!

Thanks again for your time and this awesome plugin, and I hope someone in the future finds this solution useful!

closed 2021-02-21 by dowensagain

Marus, 2021-02-21:
Thanks for following up with the solutions you found - great info to have here for future users.