Marus/cortex-debug #389: runToMain to a different function instead of main

ch1648, 2021-01-30:
Thanks for the great work. 

Is it possible to extend runToMain  to runToFunction which will take an arg to make it possible to break at a specific function instead of main?

Regards,
Chris

manuargue, 2021-01-31:
You can achieve this by adding custom post-start session commands in your launch.json, e.g. by setting a temporary breakpoint:
`"postStartSessionCommands": ["tb your_custom_function", "continue"]`

Is true that the entry point could be named different than `main`, and it that case I wonder what is happening with the debug session if `runToMain` was enabled in the launch config.

manuargue, 2021-01-31:
I gave it a try and I found out that if `runToMain` is enabled but `main` function doesn't exist, the debugger session never completes to launch. I opened pr #391 to fix this, pls @Marus take a look when you have some time.

Marus, 2021-01-31:
I'll include the runToMain fix for this upcoming v0.3.11 release.

While I won't have this for v0.3.11, although likely the next release - I'm thinking what probably makes sense is to replace `runToMain` with two properties `runToEntryPoint` (which is still a boolean, default to false) and `entryPointName` (which would default to `main`). 

runToMain would become a deprecated/undocumented configuration option which would set those two if it was set and the newer options not also given

manuargue, 2021-01-31:
@Marus I already played a bit with this code today, so it's not trouble for me submitting a pr with these modifications. Let me know if you are ok and I'll go ahead with this change.

Marus, 2021-01-31:
Yeah - make a PR if you've made the changes for sure.

ch1648, 2021-01-31:
Hello Marus,

I would love to, but I am just get involved in this as a user (try to setup my project using jlink on WSL).
Got stuck and checked into the source and find the root cause (for now I have to add a dummy main function to get it going).  However I am not quite familiar with the build and test process of the plug in. Thus will not be able to make a PR any time soon. Any place I can find a document on how to build and test with a modified plug in?
Regards,Chris

ch1648, 2021-01-31:
Please do.
Thanks.

ch1648, 2021-01-31:
I was thinking about two properties as well.
Like your idea on how to deprecate the runToMain property

ch1648, 2021-01-31:
That is what I found. runToMain has to be enabled. Adding postSessionCommand does not work.
Chris

ch1648, 2021-01-31:
Sorry, just noticed the email is for Marus from Manuel.

Regards,Chris
    On Sunday, January 31, 2021, 09:02:26 AM PST, Chris Huang <ch1648@yahoo.com> wrote:  
 
  Hello Marus,

I would love to, but I am just get involved in this as a user (try to setup my project using jlink on WSL).
Got stuck and checked into the source and find the root cause (for now I have to add a dummy main function to get it going).  However I am not quite familiar with the build and test process of the plug in. Thus will not be able to make a PR any time soon. Any place I can find a document on how to build and test with a modified plug in?
Regards,Chris

ch1648, 2021-02-03:
Hello Manuel,

I was able to use Cortex-Debug with Jlink for my project for a while, but now is not working any more.I tried with different setting for a couple days, but not making process. Hope I can get some advice from your guys.
I am using 0.3.11 version of the plug-in

The issue is after everything gets loaded and setup, the debugger will not run to main function
Here is the output of debugger output panelSEGGER J-Link GDB Server V6.80d Command Line Version
JLinkARM.dll V6.80d (DLL compiled Jun 26 2020 17:16:09)
Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1052xxx6A -jlinkscriptfile C:/Users/chrish/Documents/Neo3-1/debugger/evkmimxrt1050_sdram_init.jlinkscript -endian little -vd -ir -powertarget 1 -singlerun -strict -timeout 0 -nogui-----GDB Server start settings-----GDBInit file:                  noneGDB Server Listening port:     50000SWO raw output listening port: 50001Terminal I/O port:             50002Accept remote connection:      localhost onlyGenerate logfile:              offVerify download:               onInit regs on start:            onSilent mode:                   offSingle run mode:               onTarget connection timeout:     0 ms------J-Link related settings------J-Link Host interface:         USBJ-Link script:                 C:/Users/chrish/Documents/Neo3-1/debugger/evkmimxrt1050_sdram_init.jlinkscriptJ-Link settings file:          none------Target related settings------Target device:                 MIMXRT1052xxx6ATarget interface:              SWDTarget interface speed:        4000kHzTarget endian:                 little
Connecting to J-Link...J-Link is connected.Firmware: J-Link V11 compiled Jul 17 2020 16:24:07Hardware: V11.00S/N: 821001224Feature(s): GDBChecking target voltage...Target voltage: 3.38 VListening on TCP/IP port 50000Connecting to target...Connected to targetWaiting for GDB connection...Connected to 127.0.0.1Reading all registersWARNING: Failed to read memory @ address 0x00000000Received monitor command: speed 1000Target interface speed set to 1000 kHzReceived monitor command: clrbpReceived monitor command: resetResetting targetReceived monitor command: haltHalting target CPU......Target halted (PC = 0x802A99C2)Received monitor command: regsR0 = 809009C8, R1 = 00000077, R2 = 00000046, R3 = 80901000R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 80931260R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000R12= 00000000, R13= 80931260, MSP= 80931260, PSP= 00000000R14(LR) = 8029F95D, R15(PC) = 802A99C2XPSR 61000000, APSR 60000000, EPSR 01000000, IPSR 00000000CFBP 0400F000, CONTROL 4000000, FAULTMASK 00, BASEPRI F000, PRIMASK 00Received monitor command: speed 30000Target interface speed set to 15000 kHzReceived monitor command: flash breakpoints 1Flash breakpoints enabledReceived monitor command: semihosting enableSemi-hosting enabled (Handle on BKPT)Received monitor command: semihosting IOClient 1Semihosting I/O set to TELNET ClientReceived monitor command: clrbpReceived monitor command: resetResetting targetReceived monitor command: haltHalting target CPU......Target halted (PC = 0x802A99C2)Received monitor command: regsR0 = 809009C8, R1 = 00000077, R2 = 00000046, R3 = 80901000R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 80931260R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000R12= 00000000, R13= 80931260, MSP= 80931260, PSP= 00000000R14(LR) = 8029F95D, R15(PC) = 802A99C2XPSR 61000000, APSR 60000000, EPSR 01000000, IPSR 00000000CFBP 0400F000, CONTROL 4000000, FAULTMASK 00, BASEPRI F000, PRIMASK 00Received monitor command: haltHalting target CPU......Target halted (PC = 0x802A99C2)Received monitor command: resetResetting targetDownloading 8192 bytes @ address 0x80200000 - Verified OKDownloading 16144 bytes @ address 0x80202000 - Verified OKDownloading 16160 bytes @ address 0x80205F10 - Verified OKDownloading 16112 bytes @ address 0x80209E30 - Verified OKDownloading 16144 bytes @ address 0x8020DD20 - Verified OKDownloading 16080 bytes @ address 0x80211C30 - Verified OKDownloading 16080 bytes @ address 0x80215B00 - Verified OKDownloading 16192 bytes @ address 0x802199D0 - Verified OKDownloading 16144 bytes @ address 0x8021D910 - Verified OKDownloading 16112 bytes @ address 0x80221820 - Verified OKDownloading 16144 bytes @ address 0x80225710 - Verified OKDownloading 16176 bytes @ address 0x80229620 - Verified OKDownloading 16160 bytes @ address 0x8022D550 - Verified OKDownloading 16192 bytes @ address 0x80231470 - Verified OKDownloading 16112 bytes @ address 0x802353B0 - Verified OKDownloading 16064 bytes @ address 0x802392A0 - Verified OKDownloading 16080 bytes @ address 0x8023D160 - Verified OKDownloading 15904 bytes @ address 0x80241030 - Verified OKDownloading 16032 bytes @ address 0x80244E50 - Verified OKDownloading 16112 bytes @ address 0x80248CF0 - Verified OKDownloading 16160 bytes @ address 0x8024CBE0 - Verified OKDownloading 16176 bytes @ address 0x80250B00 - Verified OKDownloading 16064 bytes @ address 0x80254A30 - Verified OKDownloading 16144 bytes @ address 0x802588F0 - Verified OKDownloading 16112 bytes @ address 0x8025C800 - Verified OKDownloading 16176 bytes @ address 0x802606F0 - Verified OKDownloading 16176 bytes @ address 0x80264620 - Verified OKDownloading 16192 bytes @ address 0x80268550 - Verified OKDownloading 16192 bytes @ address 0x8026C490 - Verified OKDownloading 16192 bytes @ address 0x802703D0 - Verified OKDownloading 16096 bytes @ address 0x80274310 - Verified OKDownloadi [... cut ...] 1000",                "monitor clrbp",                "monitor reset",                "monitor halt",                "monitor regs",                "monitor speed 30000",                "monitor flash breakpoints 1",                "monitor semihosting enable",                "monitor semihosting IOClient 1",                "monitor clrbp",                "monitor reset",                "monitor halt",                "monitor regs",            ],                "runToMain": true,
        }    ]}

manuargue, 2021-02-04:
Hi @ch1648, in general when reporting issues I'd recommend you to attach the logs (if it's a considerable big log) or keep only the relevant bits `formatted as code`, because it may gets difficult to read.

Regarding your issue, I don't have much experience with J-Link debuggers but what I can see is that with Cortex-Debug you are getting a `WARNING: Failed to read memory @ address 0x00000000`, which may indicate that the CPU is running at the time the debugger is attempting to read the memory. Anyway it doesn't seem, at least to me, related with `runToMain` so I'd recommend to log this issue separately and perhaps @Marus can take a better look at it and help you with this issue.

ch1648, 2021-02-05:
Hi Manuel and @Marus

Thanks for getting back to me.  You are correct,  debugger cannot halt the CPU.   
I compared the Server logs from Eclipse Jink CDT vs Coretex-Debug.
Turns out Coretex-Debug make call "monitor reset" after code downloading which triggers Jlink GDB server to call jlinkscript again, this will init the sdram again.

Is there anyway to avoid this additional "monitor reset" call? or maybe monitor reset should not trigger a sdram init?

Regards,Chris

02-00000000-00-00007779-004E: T0EBC 007:781.969   Data:  FF 00 00 00 00 40 18 40 00 C2 01 00 00 00 00 00 ...02-00000000-00-00007779-0028: T0EBC 007:782.017 - 6.282ms returns 0x0003-00000000-00-00007779-000F:  - Verified OK
01-000002CC-00-00007780-0006: $OK#9a00-000002CC-00-00007780-000F: $Pf=bcaf2080#4902-00000000-00-00007780-0036: T0EBC 007:782.649 JLINK_WriteReg(R15 (PC), 0x8020AFBC)02-00000000-00-00007780-0028: T0EBC 007:782.664 - 0.016ms returns 0x0003-00000000-00-00007780-0023: Writing register (PC = 0x8020afbc)
01-000002CC-00-00007780-0006: $OK#9a00-000002CC-00-00007781-0014: $qRcmd,7265736574#3703-00000000-00-00007781-0020: Received monitor command: reset
02-00000000-00-00007781-002B: T0EBC 007:783.624 JLINK_ResetPullsRESET(ON)02-00000000-00-00007781-001B: T0EBC 007:783.637 - 0.013ms02-00000000-00-00007781-001F: T0EBC 007:783.648 JLINK_Reset()02-00000000-00-00007781-0027: T0EBC 007:783.662   ResetTarget() start02-00000000-00-00007781-0040: T0EBC 007:783.672    J-Link Script File: Executing ResetTarget()02-00000000-00-00007781-0030: T0EBC 007:783.683   J-Link script: ResetTarget()02-00000000-00-00007782-0023: T0EBC 007:784.198   DP0: 0x0BD11477

Marus, 2021-02-10:
Hi @ch1648 - Yes - we were recently made aware of some issues when debugging RAM hosted code instead of flash hosted. I've never personally done anything where I've placed the code in RAM, so I hadn't run into it until it was brought up. 

I'm going to look and see if removing that second reset has any negative effects; it seems like I had some issues with a lot of devices where it didn't pick up properly on the new stack base and/or reset vector from the vector table without the reset. Might end up needing to be a option that you can disable that second reset if you're programming to RAM instead of flash storage.

ch1648, 2021-02-10:
Hi @Marus

Thanks for getting back to me on thi.

It appears we will need to handle both.

The Jlink CDT for Eclipse has an option for RAM application.
I compare the difference for with and without this option.

With it. 
The debugger reset the device first and load the code after that it will do a monitor regs and start the debugger, (Debugger will stop at __start  or main).

Without it
The debugger will halt the device, load the code and do a reset,  this will cause the same issue as in Vcode.   (Debugger will not be able to halt CPU).

I attached two logs from JlinkGDBServerCL.exe  
log_without_ram_app_check.txtlog_with_ram_app_check.txt

For now,  I reload the code again in postLaunchCommands
            "postLaunchCommands": [                "monitor clrbp",                "monitor reset",                "monitor halt",                "monitor regs",                "file /mnt/c/Users/chrish/Documents/Neo3-1/nuttx/nuttx",                "load",                "monitor halt",                "tb __start",//                "continue",            ],            "runToMain": true,

This will make the debugger stop at "main",   but it will load the code twice which is time consuming.

-Chris

 
 

  
SEGGER J-Link GDB Server V6.80d LogFile
Logging started @ 2021-02-10 21:32
03-00000000-00-00000000-0136: Command line: -if swd -device MIMXRT1052xxx6A -endian little -speed 30000 -port 2331 -swoport 2332 -telnetport 2333 -vd -ir -localhostonly 1 -powertarget 1 -singlerun -strict -timeout 0 -nogui -log c:\temp\log.txt -jlinkscriptfile C:\Users\chrish\Documents\Neo3-1\debugger\evkmimxrt1050_sdram_init.jlinkscript

03-00000000-00-00000000-01B1: -----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Generate logfile:              on

03-00000000-00-00000002-019D: ------J-Link related settings------

03-00000000-00-00000002-0001: 

03-00000000-00-00000002-0018: Connecting to J-Link...

02-00000000-00-00000006-0046: TB540 000:009.180   Firmware: J-Link V11 compiled Jul 17 2020 16:24:07
02-00000000-00-00000007-003A: TB540 000:009.374   Decompressing FW timestamp took 140 us
02-00000000-00-00000011-0024: TB540 000:012.669   Hardware: V11.00
02-00000000-00-00000011-0022: TB540 000:012.695   S/N: 821001224
02-00000000-00-00000011-001F: TB540 000:012.707   OEM: SEGGER
02-00000000-00-00000011-0023: TB540 000:012.717   Feature(s): GDB
02-00000000-00-00000012-003F: TB540 000:014.768   TELNET listener socket opened on port 19021
02-00000000-00-00000012-002D: TB540 000:014.921   WEBSRV Starting webserver
02-00000000-00-00000012-0040: TB540 000:015.081   WEBSRV Webserver running on local port 19080
02-00000000-00-00000012-002B: TB540 000:015.097 - 10.803ms returns "O.K."
03-00000000-00-00000012-0015: J-Link is connected.

02-00000000-00-00000013-0043: TB540 000:015.201 JLINK_ExecCommand("HideDeviceSelection=1", ...). 
02-00000000-00-00000013-0028: TB540 000:015.229 - 0.006ms returns 0x00
02-00000000-00-00000013-0043: TB540 000:015.242 JLINK_ExecCommand("DisableInfoWinFlashDL", ...). 
02-00000000-00-00000013-0028: TB540 000:015.256 - 0.001ms returns 0x00
02-00000000-00-00000013-0044: TB540 000:015.268 JLINK_ExecCommand("DisableInfoWinFlashBPs", ...). 
02-00000000-00-00000013-0028: TB540 000:015.281 - 0.001ms returns 0x00
02-00000000-00-00000013-003B: TB540 000:015.295   JLINK_SetFlashProgProgressCallback(...)
02-00000000-00-00000013-001E: TB540 000:015.305     returned
02-00000000-00-00000013-0046: TB540 000:015.317 JLINK_ExecCommand("device = MIMXRT1052xxx6A", ...). 
02-00000000-00-00000020-005B: TB540 000:022.219   XML file found at: C:\Program Files (x86)\SEGGER\JLink\JLinkDevices.xml
02-00000000-00-00000020-0060: TB540 000:022.958   C:\Program Files (x86)\SEGGER\JLink\JLinkDevices.xml evaluated successfully.
02-00000000-00-00000063-0036: TB540 000:065.607   Device "MIMXRT1052XXX6A" selected.
[... 809 lines cut ...]
02-00000000-00-00002758-0026: T693C 002:760.327 JLINK_ReadReg(FPS17)
02-00000000-00-00002758-002E: T693C 002:760.336 - 0.009ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.346 JLINK_ReadReg(FPS18)
02-00000000-00-00002758-002E: T693C 002:760.355 - 0.009ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.366 JLINK_ReadReg(FPS19)
02-00000000-00-00002758-002E: T693C 002:760.375 - 0.009ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.385 JLINK_ReadReg(FPS20)
02-00000000-00-00002758-002E: T693C 002:760.394 - 0.009ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.405 JLINK_ReadReg(FPS21)
02-00000000-00-00002758-002E: T693C 002:760.442 - 0.037ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.453 JLINK_ReadReg(FPS22)
02-00000000-00-00002758-002E: T693C 002:760.463 - 0.010ms returns 0x00000000
02-00000000-00-00002758-0026: T693C 002:760.488 JLINK_ReadReg(FPS23)
02-00000000-00-00002759-002E: T693C 002:760.518 - 0.051ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.551 JLINK_ReadReg(FPS24)
02-00000000-00-00002759-002E: T693C 002:760.560 - 0.009ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.571 JLINK_ReadReg(FPS25)
02-00000000-00-00002759-002E: T693C 002:760.594 - 0.023ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.624 JLINK_ReadReg(FPS26)
02-00000000-00-00002759-002E: T693C 002:760.656 - 0.032ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.705 JLINK_ReadReg(FPS27)
02-00000000-00-00002759-002E: T693C 002:760.714 - 0.009ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.725 JLINK_ReadReg(FPS28)
02-00000000-00-00002759-002E: T693C 002:760.752 - 0.027ms returns 0x00000000
02-00000000-00-00002759-0026: T693C 002:760.763 JLINK_ReadRe

Marus, 2021-02-13:
The new `runToEntryPoint` configuration property has been release with v0.3.12.

@ch1648 - Can you please open a new issue to cover the issues you are having when loading the code to RAM so we can follow up with that issue separately, as I do think that is an important issue to try to fix soon. Please copy any of the relevant info from this thread into the new issue.

closed 2021-02-13 by Marus