Marus/cortex-debug #390: Does not play well with Settings Sync across different platforms

tntodorov, 2021-01-30:
I am forced to work with different OSes because of customer requirements and use the settings sync feature in VSCode.

Because paths differ between Windows and MacOS and Linux, currently I have to set Sync to ignore settings for cortex-debug. It would be nice if all the settings which reference a path had OS variants. Let me give an example.

Currently:
```
  "cortex-debug.JLinkGDBServerPath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
```

What I would like to have:
```
  "cortex-debug.JLinkGDBServerPath": "/opt/JLink/JLinkGDBServerCL",
  "cortex-debug.JLinkGDBServerPath.windows": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
```

The standard `cortex-debug.JLinkGDBServerPath` should still be there for backwards compatibility and people that work exclusively with a single OS. Then there are the variants `....linux`, `....osx`, and `....windows` which overwrite the default on the corresponding system. People should be able to use only default, mixed default and OS variants, or only OS variant settings in their settings.json file. This should be implemented for all paths - toolchain, OpenOCD, etc.

Thanks for developing this great extension!

Marus, 2021-01-31:
This is certainly a good improvement. I'll look into this and hopefully can include it in a release in the not-to-distant future (although it won't be in V0.3.11 that should hopefully be released today).

tpilt, 2021-02-28:
There's a relation to 
https://github.com/Marus/cortex-debug/issues/401
here.

I am achieving the same in my tasks.json by using an environment variable (which is not possible currently in settings.json etc). This is then set differently on different OS'es, and on the different developers machines. 

Just because it is windows, does not mean, that everybody chose same install-dir.

haneefdm, 2021-12-22:
For every path mentioned for user/workspace settings, there is now a OS-specific setting. This will work better for users using the sync feature. This is also the method used by VSCode itself. I recommend using this mechanism rather than `serverpath` to keep things portable. From CHANGELOG.md for V0.4.8

> Pathnames for gdb-servers can be OS specific. For instance cortex-debug.openocdPath can be suffixed with one of .linux, .osx or .windows. For instance cortex-debug.openocdPath.windows is used only on Windows and if that is missing, it will default looking for cortex-debug.openocdPath`.

closed 2021-12-22 by haneefdm