Marus/cortex-debug #396: Failed to lunach ST-Link GDB Server: Error: spawn c./....ST-LINK_gdbserver.exe ENOENT

dm5xx, 2021-02-06:
Hi,

i am using this launch.config:
{
    "version": "0.2.0",
    "configurations": [
        
        {​​​​​​​​
            "cwd": "${​​​​​​​​workspaceRoot}​​​​​​​​",
            "executable": "C:\\Projects\\mbed-projects\\BUILD\\F411\\Debug\\F411.elf",
            "name": "debug stlink",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "stlink",
        }​​​​​​​​
    ]
}

Starting a debugging session brings this output in the debugging console:

Please check OUTPUT tab (Adapter Output) for output from c:\ST\STM32CubeIDE_1.5.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.5.0.202011040924\tools\bin\ST-LINK_gdbserver.exe

Launching server: "c:\ST\STM32CubeIDE_1.5.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.5.0.202011040924\tools\bin\ST-LINK_gdbserver.exe" "-p" "50000" "-cp" "c:\ST\STM32CubeIDE_1.5.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_1.5.0.202011040924\tools\bin" "--swd"

VS Code Error windows says: Failed to launch ST_LINK GDB Server: Error: spawn c.\ST\....\com-st-stm32cu...\ST-LINK_gdpserver.exe ENOENT

Hint: the first "..." above are not really shown as ... - just just beacause its the lon default path. the "stm32cu..." is real.

When im running:
c:\ST\STM32CubeIDE_1.5.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.5.0.202011040924\tools\bin\ST-LINK_gdbserver.exe -p 50000 -cp c:\ST\STM32CubeIDE_1.5.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_1.5.0.202011040924\tools\bin --swd

STMicroelectronics ST-LINK GDB server. Version 5.7.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J37S7
Device ID: 0x431
PC: 0x80045d0
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.17 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Waiting for debugger connection...
Waiting for connection on port 50000...
Stopping port 50000
Received stop for port 50000, not accepting new connections.

The server is running. I am using a stm32F411ceu connected to a stlink v2 china clone with newest firmware. I have no idea whats going wrong.

thanks for your help! :)
Mike.

manuargue, 2021-02-06:
Hi. I assume you have STM32Cube related tools set in your Path environment variable already and Cortex-Debug is automatically resolving the paths. Have you tried to override these from the launch settings using `armToolchainPath`, `stm32cubeprogrammer` and `serverpath`?

dm5xx, 2021-02-06:
hi manu, yes, i tried overriding these settings - no success. i pointed all 3 settings to the used paths. 
but maybe i am missing something in the path environment?

i tried a lot, but im not sure. so can we step back 1 step and please shortly gimme some examples how the path env variable and the 3 paths should look alike. stm32cubeide is in the default location c:/st/...

M.

hwmaier, 2021-02-06:
Are you sure that the quoted output is correct:

> VS Code Error windows says: Failed to launch ST_LINK GDB Server: Error: spawn c.\ST....\com-st-stm32cu...\ST-LINK_gdpserver.exe ENOENT

Because this makes no sense at all. The `C:` became a `c.` and the executable name would be wrong, please note the `p` in `gdpserver.exe`.

hwmaier, 2021-02-07:
Also when you configure `armToolchainPath`, `stm32cubeprogrammer` etc you are using the path separator wrong. It muts be either `\\` or `/`. A single backslash won't work. I believe this is also true for `executable`. 

But it should work without these overrides and you don't need anything in the path for the stlink mode. The plugin should autodiscover the location of the tools involved.

hwmaier, 2021-02-07:
I think I just saw the reason, you are using STM32CubeIDE_1.5.1. The plugin was developed against STM32CubeIDE_1.5.0 and 1.40. And maybe ST changed the directory structure or something. Can you try against STM32CubeIDE_1.5.0  and see if that works for you?

manuargue, 2021-02-07:
Sorry for late reply. I correct myself, there’s no need to have it in the Path env variable, but only to keep the default installation directory which in your case seems to be correct.

Pls pay attention to the slashes as @hwmaier mentioned above and let us know if that went ok.

dm5xx, 2021-02-07:
Hi and good morning,

Sorry for the confusion about the path. They are \\ but this is not shown in the thread here. I also tested yesterday with /.  There is no difference. Using wrong slashes immediately caused a different error. So the slashe unfortunately are not the problem. Also changing the executable path (to something bin existing) resulted immediately in a different error 

So let's assume the path are correct with \\. 

Mike

dm5xx, 2021-02-07:
So I will now follows hendriks advice to use version 1.5.0. i also will change all slashes to / so there is no problem if i post some thing here.

M.

dm5xx, 2021-02-07:
HAHA! Thank you Hendrik, i found the problem:

----> "cwd": "${workspaceFolder}",

and not: "cwd": "${workspaceRoot}",

Also no need to downgrate to 1.5.0 and also no need to override the 3 other values... But i no do have another problem...

Not implemented stop reason (assuming exception): undefined
STM32 Successfully completed reset operation
Error finishing flash operation
Remote connection closed

And the plugin says:
Failed to launch GDB: Remote connection closed (from interpreter-exex console "monitor reset halt"

Maybe i should open another issue for that?

Mike.

manuargue, 2021-02-07:
I’m glad you were able to make it work. 

Regarding the flashing error, I’ve noticed you mentioned you are using stlink v2 with latest firmware but in your settings you are passing  `“v1": true`. This may be correct depending on the version, but since you are using a clone is difficult to tell. Just in case, give it a try setting it to false.

closed 2021-08-16 by haneefdm

Zhang-bingrui, 2024-05-04:
Hi I'm using this lauch.config
I get the problem as well ,  I check all settings as what you have seen , but I can't solve this problem. I also removed the stlink_gdbserver and instaledl it agin,but it doesn't work.My setting jason is 
{
    
    "version": "0.2.0",
    "configurations": [

        {
            "cwd": "${workspaceFolder}",
            "executable": "./build/SmartGreenhouses_Test/SmartGreenhouses_Test.elf",
            "name": "Debug with ST-Link",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
            "servertype": "stlink"
        }

    ]
}
Could you give me a suggestions?