Marus/cortex-debug #398: Unable to set breakpoints since VS Code update 1.53

richardtearle-nps, 2021-02-10:
Hi

I've been unable to set breakpoints in the VS code UI since the VS Code update 1.53. The recent 1.53.1 update to VS code hasn't resolved this.

I'm using ST Link v2 connecting to STM32F042 using openocd version 0.10.0-13.

Debugging works, and I can set breakpoints with the thumb instruction BKTP (and rebuilding the code/etc...)

Marus, 2021-02-12:
What host OS and version of cortex-debug are you using?

Also, can you try running with `showDevDebugOut` set to true? Post any output as an attachment please as it is just about impossible to read when posted inline.

richardtearle-nps, 2021-02-12:
Hi, it's windows 10 (64-bit - version 2004 [Build 19041.746]), and version 0.3.11 of cortex-debug

Where do I set `showDevDebugOut`? I've added it to my `settings.json` (and added `cortex-debug.showDevDebugOut`), but I don't see any debug output. Sorry.

Marus, 2021-02-12:
Sorry, that should be a parameter in your launch.json configuration that you are using. Should just give extra output. 

What language are you developing in? Just wondering if it is language specific that it can’t set breakpoints for. 

I’ll try getting things going on my windows 10 machine to see if I can replicate any similar issues - I’m not noticing any issues on my macOS or Linux machines that I typically use.

perlindgren, 2021-02-12:
I have the same problem under Linux, I can't set a breakpoint in the code view (since the update) neither by clicking on toggling with F9 (but I don't know if toggling worked before, never used that...).
/Per

Marus, 2021-02-12:
@perlindgren - I think you do a lot of rust stuff if I recall. Do you happen to have a c example to try - to see if it is language specific (even though rust is listed as a supported language for the debugger).

richardtearle-nps, 2021-02-12:
(Don't laugh....) I writing straight assembler/thumb2 code. Still built with arm-none-eabi-gcc toolchain.

Sorry I'm making heavy weather of the `showDevDebugOut` setting, but I get `"Property showDevDebugOut is not allowed."` when launching, when I add it to my launch.json file.

perlindgren, 2021-02-12:
> @perlindgren - I think you do a lot of rust stuff if I recall. Do you happen to have a c example to try - to see if it is language specific (even though rust is listed as a supported language for the debugger).

Nope, no C code since years :) I would be surprised if it would be language dependent, but on the other hand, tooling is full of surprises ;)

Marus, 2021-02-12:
While it does seem unlikely - you are both not using C/C++ (which is what my stuff is in). I have a feeling that for some reason VS code just isn't giving the debug adapter the proper information to set the breakpoints or something - as I don't see it being an interaction between the extension - maybe it's not getting registered properly as being able to handle debugging for those languages (why that changed with this version I don't know).

I'll try to get a simple rust example going for one of my boards so I can try that out.

What is the language (and associated language ID) showing in the bottom status bar for the files you are trying to set the breakpoints in?

richardtearle-nps, 2021-02-12:
Mine's set as ARM (id: arm) - I have the [ARM](https://github.com/dan-c-underwood/vscode-arm.git) plugin (version 1.5.0) installed as well.

If I change the language to C then I can set breakpoints again.

Marus, 2021-02-12:
Yeah, I can try to add that identifier to the list of supported languages- which should fix it. I’m more confused about @perlindgren though - as “rust” is listed as supported (unless there is a new Rust extension that changes the language ID). Not sure why it started caring about it all of a sudden though - must be some change in VSCode.

Marus, 2021-02-12:
Ah, I think I’m seeing something now (arm was actually already listed). 

There was a change to some of the structure in the package.json manifest and I think they dropped the old way of doing it completely now. I’ll upload a pre-release for you to try shortly.

Marus, 2021-02-12:
@perlindgren @richardtearle-nps - There is a new v0.3.12-pre1 release on the GitHub releases page - can you try to install that .vsix file instead of from the marketplace and see if it fixes the issue with not being able to set breakpoints (with your language set correctly). If that fixes the issue then I'll try to release that later today.

richardtearle-nps, 2021-02-12:
Works for me :-D

Thanks.

Marus, 2021-02-12:
Sounds good - will do the final v0.3.12 release tonight and then you can go back to the marketplace version.

Marus, 2021-02-13:
I just release v0.3.12 to the marketplace - which I think will fix the issues with setting breakpoints both with assembly (including the disassembly views) and rust. If you still have problems after the upgrade please re-open the issue.

closed 2021-02-13 by Marus

perlindgren, 2021-02-14:
I made a simple test and it works. (Did not test assembly level breakpoints though...)
Great work!
/Per