Marus/cortex-debug #399: Not implemented stop reason (assuming exception): undefined

TarikAkyuz, 2021-02-11:
I get the warning below.
```
63	  ldr   r0, =_estack
Not implemented stop reason (assuming exception): undefined
```

.vscode/launch.json

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "./build/Led-blink.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32G431RB",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32g4x.cfg"
            ],
            "svdFile": "./STM32G431xx.svd"
        }
    ]
}
```

.vscode/c_cpp_properties.json

```
{
    "configurations": [
        {
            "name": "Nucleo LedBlink",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE",
                "USE_HAL_DRIVER",
                "STM32G431xx"
            ],
            "intelliSenseMode": "linux-gcc-arm",
            "compilerPath": "/opt/st-toolchain/bin/arm-none-eabi-gcc",
            "compilerArgs": [
                "-mcpu=cortex-m4",
                "-mthumb"
            ],
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```

[image]

full workspace: [https://github.com/TarikAkyuz/vs-debug](url)

haneefdm, 2021-07-10:
Sorry, it took us this long for us to see this. That is actually normal behavior with most embedded environments using gdb. Gdb does not know why the program stopped (at the reset vector) only that it did stop. The gdb-server is supposed to report a reason (breakpoint, exception, SIGNAL, etc.) which in this case it doesn't report anything.

We just got used to it. We just report what gdb and the gdb-server (OpenOCD, PyOCD, JLink, etc.) report

haneefdm, 2021-07-10:
Your program will continue to work normally if you resume.

You can also use `"runToEntryPoint": "main"` in your `launch.json` to skip stopping at the reset handler (the default that does not waste a breakpoint). You can change `main` to any function name you want.

closed 2021-07-10 by haneefdm