Marus/cortex-debug #401: Add environment variables support for both settings and launch config files (feature request)

tdjastrzebski, 2021-02-20:
Please consider the following feature: support for environment variables in config files (**settings.json** & **launch.json**) done the same way it is implemented in [vscode-cpptools](https://code.visualstudio.com/docs/languages/cpp) for VSCode using "**env:**" variable prefix.

Example:
`"cortex-debug.openocdPath": "{env:OpenOCD_PATH}/bin"`
`"searchDir": ["{env:OpenOCD_PATH}/scripts"]`

Implementation probably can be borrowed from [cpptools resolveVariables() function](https://github.com/microsoft/vscode-cpptools/blob/256d0cd95feda81514ae18a16f396ee73dcfb280/Extension/src/common.ts#L303).

tpilt, 2021-02-28:
That would be really helpful. 
1- People have tools installed differently.
2 - projects will over the years depend on tools in different versions. Therefore it will be nice to be able to specify the right tools using e.g. 
"cortex-debug.openocdPath": "{env:OpenOCD_V4_PATH}/bin"
(notice the V4)
3 - Consistency - I was very surprised that the rules in the json files was different from tasks.json (owned and parsed by MS c_cpp plugin ?) and the launch.json. (But I know: they also added input variables etc - can't be up to date with everything they throw in that plugin ...)

haneefdm, 2021-07-10:
@tdjastrzebski, @tpilt  Ummm, Two very different things, and we don't make the rules!

* 'tasks.json' and 'launch.json' already support env. variables and a host of other ways to do substitutions. These are supported for all VSCode debuggers.
* `settings.json` does not and is an incomplete file. It also does not support the os-specific sections. VSCode manages this and combines user, workspace & folder settings into one set of settings for it. It has no facilities like that. So, what you see in your local settings.json is only partial info.

None of these are even true JSON files (note: JSON does not allow comments)

What is mentioned in #390 is something we can implement. Even VSCode does this for the shell path. Or we may use the cpptools way of doing it but that is very error prone. Either way, it is a bunch of busy work.

[image]

Until we have a release with this there are already (good) workarounds that you can use.
* Use user-level settings to set the paths. These are user/os specific anyways and it doesn't matter if each user in the team installs stuff in a different location or use different OSes
* Use environment variables and use `armToolchainPath` and `serverpath` in launch.json to refer to those.

haneefdm, 2021-12-22:
For every path mentioned for user/workspace settings, there is now a OS-specific setting. This will work better for users using the sync feature. This is also the method used by VSCode itself. I recommend using this mechanism rather than `serverpath` to keep things portable. From CHANGELOG.md for V0.4.8

> Pathnames for gdb-servers can be OS specific. For instance cortex-debug.openocdPath can be suffixed with one of .linux, .osx or .windows. For instance cortex-debug.openocdPath.windows is used only on Windows and if that is missing, it will default looking for cortex-debug.openocdPath`.

closed 2021-12-22 by haneefdm