Marus/cortex-debug #403: Minor annoyance: "Restart" button also recompiles the code when there is a "preLaunchTask"

TorBlackbeard, 2021-02-27:
Minor annoyance:
If my launch configuration has a "preLaunchTask" (to build the code), that will be executed as well, when pressing "Restart" debugger button. 
This can be annoying, if my build is slow. I just want to restart debugging the exact same elf file again.

Is there another way to compile the code before the launch, but *not* when pressing Restart?

Marus, 2021-03-05:
Unfortunately “preLaunchTasks” are something that is handled by VSCode and not by the cortex-debug extension - so I don’t think we have any way to customize that behaviour. I’d suggest searching in general VSCode places to come up with a solution

tpilt, 2021-03-07:
OK, thx for attention - just close issue

trond-snekvik, 2021-07-28:
How about creating a proprietary "restart" button on the toolbar to resolve this? This would also resolve this comment, and let the restart button work the way it does for other debuggers: 

https://github.com/Marus/cortex-debug/blob/6a262b544ca49baa728789da3225b11b368fa3be/src/gdb.ts#L827-L834

This could be achieved by contributing a menu item for the "debug/toolbar" menu context, and could look like this, for instance:

[image]

haneefdm, 2021-07-28:
There is this whole "Debug Protocol" that we have to deal with. While we can hack something together, I don't recommend that. Btw, that was me that added that comment you quote :-)

Despite my comments, in the desktop (or even otherwise) world, what Microsoft said actually is right. When you detach, the program is supposed to continue running and re-attach will find the program in a different spot. But most (all) of the gdb-servers in the embedded space implement detach wrongly

You can probably write a gdb function that you can run to do the same... perhaps bind that to a key.?

haneefdm, 2021-07-28:
I also recommend you file an issue with Microsoft VSCode. They are generally responsive. I do agree that a restart should not need a pre-launch task. But now, they will have a compatibility issue

haneefdm, 2021-09-08:
Looking into the 'reset device' button as suggested by @trond-snekvik

haneefdm, 2021-09-09:
In the next release, there will be a new `Reset device` button in the debug toolbar. Microsoft should have fixed the 'Restart' button but very unlikely.

TorBlackbeard, 2021-09-10:
I wrote an issue on this on MS VSCode repo, but got no response.
Excelent, I'm looking forward to this!

closed 2021-09-11 by haneefdm