Marus/cortex-debug #408: Halt after stop debugging the MIMRT6xx process

v-rakkun, 2021-03-09:
Hello Team,

I using cortex-debug extension to debug RT6xx series processor. It is working fine for debugging. but once debugging is stopped, CPU going to halt state. May I know the setting to keep CPU is in running state after stop debugging.

Thank you
Rakesh K

oswinrodrigues, 2021-03-09:
This seems similar to #348

haneefdm, 2021-07-10:
There are two scenarios
1. Launch type debug session: When you stop, the program will be in a halted state. This is what gdb does.
2. Attach type debug session: When you stop, you are actually detaching and we forward the request to gdb and the gdb-server. The gdb-server is supposed to do the actual resume but some servers violate that understanding.

If you are using OpenOCD and using an 'attach' type session, then chances are it will fail to resume and leave the program in a halted state (very bad). Not much we can do. See also #348

closed 2021-07-10 by haneefdm