Marus/cortex-debug #409: Need required settings for hex format view in watch and variables section

v-rakkun, 2021-03-09:
Hello Team,

I using cortex-debug extension to debug RT6xx series processor. It is working fine for debugging. In watch able to see the varibales in hex format by adding ,x to the variable name. Instead of adding it to every variable, is there any global setting to see variables and watch window values in HEX format by default ?

Thank you
Rakesh K

Marus, 2021-03-09:
No - for watch variables they are displayed in the GDB default format for the type of data; there is currently no global setting to override this.

bhargavshah1988, 2021-03-09:
@Marus what are your thoughts to add a configuration for this ?

FabianoGK, 2021-03-19:
You could add `set output-radix 16` to the `postLaunchCommands`.

v-rakkun, 2021-03-29:
Thank you @FabianoGK.

Post launch command(set output-radix 16) is working as expected.

closed 2021-03-29 by v-rakkun