Marus/cortex-debug #412: OpenOCD server drops GDB connection with LIBUSB error

oswinrodrigues, 2021-03-09:
### Steps To Reproduce

1. Start debug session, arrive at `main` entry
2. Set breakpoint a couple lines down
3. Continue code-execution until said breakpoint
4. GDB connection to OpenOCD server drops
5. OpenOCD log shows a `libusb_handle_events() failed with LIBUSB_ERROR_INTERRUPTED` message

### Setup

<details><summary>launch.json - Click To Expand</summary>

```json
{
    "name": "(gdb) Stack Controller",
    "type": "cortex-debug",
    "request": "launch",
    "cwd": "${workspaceFolder}",
    "servertype": "openocd",
    "serverArgs": ["-c ftdi_serial nebb13"],
    "searchDir": ["${workspaceFolder}/prog"],
    "configFiles": ["sc_busblaster.cfg"],
    "device": "TM4C1294NCPDT",
    "executable": "${workspaceFolder}/fw/bin/tiva/sc.exe",
    "runToEntryPoint": "main",
    "overrideLaunchCommands": ["monitor reset halt"]
}
```
</details>

I have an interesting networking setup. Using VS Code on my Mac at home. With the _Remote - SSH_ extension, I tunnel into an Ubuntu 18.04 VM that is hosted on my Windows 10 machine at work. The `openocd` and `gdb` installations are on this VM. As well, the debug probes' USB connections are captured by the VM (physically connected to the machine at work). I tried removing some of these layers by launching a debug session directly on the VS Code instance in the VM, and I see the same behaviour.

### Logs

<details><summary>OpenOCD - Click To Expand</summary>

```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
trst_only separate trst_push_pull
adapter speed: 500 kHz
Info : clock speed 500 kHz
Info : SWD DPIDR 0x2ba01477
Info : tm4c1294ncpdt.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/50000
undefined debug reason 7 - target needs reset
adapter speed: 500 kHz
cortex_m reset_config sysresetreq
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000177c msp: 0x200014cc
Info : dropped 'gdb' connection
Error: libusb_handle_events() failed with LIBUSB_ERROR_INTERRUPTED
```
</details>

<details><summary>GDB - Click To Expand</summary>

```
Please check OUTPUT tab (Adapter Output) for output from openocd
Launching server: "openocd" "-c" "gdb_port 50000" "-s" "/home/orodrigues/bms-device/prog" "-f" "sc_busblaster.cfg" "-c ftdi_serial nebb13"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /home/orodrigues/bms-device/fw/bin/tiva/sc.exe...
done.
main () at sc/src/main.cpp:57
57	int main() {
Not implemented stop reason (assuming exception): undefined
adapter speed: 500 kHz
cortex_m reset_config sysresetreq
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000177c msp: 0x200014cc
Note: automatically using hardware breakpoints for read-only addresses.

Temporary breakpoint 1, main () at sc/src/main.cpp:57
57	int main() {
b 78
Breakpoint 2 at 0x264fe: file sc/src/main.cpp, line 78.
{"token":707,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
c
Continuing.
{"token":710,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}

Breakpoint 2, main () at sc/src/main.cpp:78
78	    if (board.getJtagDisabledState())
```
</details>

### Some Observations

- The last line in the OpenOCD log, with the LIBUSB error, does not always appear. But the behaviour is consistent.
- This behaviour is not observed if I use `openocd` and `gdb` in the terminal directly.
- For some odd reason, this problem does not happen if the breakpoint is between lines 57 and 78. Also if I place a breakpoint it other functions, at places that occur after `main.cpp:78` in the code path.
- (Might be unrelated but) sometimes, a `step` over / into / out seems to act as if it was a `continue` operation. The code just runs and doesn't stop until it reaches a breakpoint.

closed 2021-03-09 by Marus

oswinrodrigues, 2021-03-09:
> Don't have any idea why you would be getting different behaviour when running it directly within that VM

Just to clarify, this is *not* the case. Rather "I see the same behaviour" when trying to reproduce directly (using RDP) in the VM i.e. removing the whole SSH + VPN tunnel layering.

That said, I think I understand that's besides the point. The VM's USB connection-handling seems to be what you're suggesting is the issue. My question then is: in order to avoid the high bursts of traffic, **is there any way to optionally disable or control how much information is loaded?** I wouldn't need anything more than just variables in the local scope.

haneefdm, 2021-03-09:
The only VM I trust with USB is VMWare. While it costs money, it is worth the frustration. Some people had luck with Parallels, but I have not tried it

oswinrodrigues, 2021-03-09:
@Marus, any thoughts? Many thanks 🙏🏽

oswinrodrigues, 2021-03-10:
Gotcha. So to confirm, this extension isn't really meant to be used inside a VM then? It may work, it may not, depending on the VM environment, etc.? 

If so, then maybe this is something that can be flagged on the extension's landing page, that this is an undocumented and unsupported use case and behaviour? Definitely would be good for someone else in my shoes to know when considering whether to install the cortex-debug extension.

On that note, seeing that this was based on the [_Native Debug_](https://marketplace.visualstudio.com/items?itemName=webfreak.debug) extension, does anyone know if that happens to have VM support?

oswinrodrigues, 2021-03-11:
@Marus, any thoughts?

oswinrodrigues, 2021-03-12:
I confirmed this works on a host OS. So yes, this is an issue with the VM's USB handling.

@Marus, I'll leave it to you to decide whether to mention this on the README / the [extension's landing page](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug). It's not uncommon for developers to be working within a VM and so this might be a good thing to flag.

oswinrodrigues, 2021-03-30:
> On that note, seeing that this was based on the [Native Debug](https://marketplace.visualstudio.com/items?itemName=webfreak.debug) extension, does anyone know if that happens to have VM support?

@Marus, would you know anything about this? Thanks for your time 🙏🏽