Marus/cortex-debug #413: add ability to choose 'mps2-an385' machine

TorBlackbeard, 2021-03-11:
It would be very helpful to add ability to choose the "mps2-an385" machine here:
https://github.com/Marus/cortex-debug/blob/35f42186726063bcb150f2e4132556a7528eea60/package.json#L303
It's just a cortex-m3, but it has lots of flash and ram. 

Just allow user to choose "msp2-an385". Or make it possible to choose "", and then user can add whatever he wants via the "serverArgs". Maybe he has recompiled it himself, or he has installed the xPack version with all the stm32 models. 

When running unittest in qemu, 64 kb ram disappears quickly.
Simple examples for the 'lm3s6965evb' runs unmodified on msp2-an385, including semihosting features.

TorBlackbeard, 2021-03-11:
https://github.com/Marus/cortex-debug/pull/414

TorBlackbeard, 2021-03-11:
By the way, I tried to locate the package.json file in my filesystem, and ninja-edited it manually. Just to try it out....
... but my newly added string did not appear in the dropdown ...

How would I test this myself before an official release?

Marus, 2021-03-11:
The PR I think looks fine, so I’ll include in the next release. 

I expect it will still work fine even if you put in the value currently despite it not being in the package.json - that pretty much just provides the hints when editing the launch.json. You may get some warning in the launch.json file, but expect it will work properly as I intentionally didn’t validate that value in the JavaScript code.

TorBlackbeard, 2021-03-12:
Oh, of cause, works!
`Launching server: "qemu-system-arm" "-cpu" "cortex-m3" "-machine" "mps2-an385" `

haneefdm, 2021-07-10:
PR #414

closed 2021-07-10 by haneefdm