Marus/cortex-debug #415: Unable to copy some values

diggit, 2021-03-12:
Hi,
There seems to be something broken in copying  (righ click -> copy value) of some values from Watch View.

Instead of value, error is copied, eg.:
`
<There is no member or method named protected. (from var-create watch_8013f2390c0b3cf472c03c0bf0501ace3e4bb638a5e114163b69c9cc00fa8c63 @ "txQueue.protected.m_elements.private._M_ptr")>
`

Here is log of me expanding view of `txQueue` and trying to copy value of `txQueue.protected.m_elements.private._M_ptr`
```
225-var-list-children --all-values watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected
GDB -> App: {"token":225,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["numchild","7"],["children",[["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_name"],["exp","m_name"],["numchild","1"],["value","{...}"],["type","Ar::Name"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_alive"],["exp","m_alive"],["numchild","2"],["value","{...}"],["type","std::atomic<bool>"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements"],["exp","m_elements"],["numchild","2"],["value","{...}"],["type","std::span<std::aligned_storage<1, 1>::type, 4294967295>"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_writePos"],["exp","m_writePos"],["numchild","2"],["value","{...}"],["type","std::atomic<Ar::queue::Mpmc<unsigned char>::TPositionWrite<unsigned short> >"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_readPos"],["exp","m_readPos"],["numchild","2"],["value","{...}"],["type","std::atomic<Ar::queue::Mpmc<unsigned char>::TPositionRead<unsigned short> >"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_pushBlockedList"],["exp","m_pushBlockedList"],["numchild","1"],["value","{...}"],["type","Ar::List"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_popBlockedList"],["exp","m_popBlockedList"],["numchild","1"],["value","{...}"],["type","Ar::List"]]]]],["has_more","0"]]}}
226-var-list-children --all-values watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements
GDB -> App: {"token":226,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["numchild","2"],["children",[["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.public"],["exp","public"],["numchild","1"],["value",""]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private"],["exp","private"],["numchild","2"],["value",""]]]]],["has_more","0"]]}}
227-var-list-children --all-values watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private
GDB -> App: {"token":227,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["numchild","2"],["children",[["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private._M_extent"],["exp","_M_extent"],["numchild","1"],["value","{...}"],["type","std::__detail::__extent_storage<4294967295>"]]],["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private._M_ptr"],["exp","_M_ptr"],["numchild","1"],["value","0x20001bb8 <fd1+2280>"],["type","std::span<std::aligned_storage<1, 1>::type, 4294967295>::pointer"]]]]],["has_more","0"]]}}
228-var-list-children --all-values watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private._M_ptr
GDB -> App: {"token":228,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["numchild","1"],["children",[["child",[["name","watch_cb9acb03a0009bdd3b6041afbfe83a1409f6a1e588fea312d2a3cd120619e584.protected.m_elements.private._M_ptr.public"],["exp","public"],["numchild","2"],["value",""]]]]],["has_more","0"]]}}
229-var-update --thread 1 --frame 0 --all-values watch_8013f2390c0b3cf472c03c0bf0501ace3e4bb638a5e114163b69c9cc00fa8c63
GDB -> App: {"token":229,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
230-var-create watch_8013f2390c0b3cf472c03c0bf0501ace3e4bb638a5e114163b69c9cc00fa8c63 @ "txQueue.protected.m_elements.private._M_ptr"
GDB -> App: {"token":230,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","There is no member or method named protected."]]}}
watch: There is no member or method named protected. (from var-create watch_8013f2390c0b3cf472c03c0bf0501ace3e4bb638a5e114163b69c9cc00fa8c63 @ "txQueue.protected.m_elements.private._M_ptr")
```

Value of pointer, I am trying to copy, is displayed in Watch View:
[image]

After adding `txQueue.m_elements._M_ptr` into Watch View (note omitted access specifiers). Copying value works as expected.

It seems that cortex-debug does not strip `private`, `public` and `protected` access specifiers from object path when querying gdb, but only when requesting value for copy, because Watch View displays value.

I am using cortex-debug 0.3.12 in vscodium

Marus, 2021-03-12:
Will look into this, hopefully shouldn’t be too hard to sort out.

haneefdm, 2021-10-09:
This is not C++-specific. It can also happen with C expressions. We (Cortex-Debug) provides VSCode with an `evaluateName` which is supposed to be an expression. We create these expressions when a parent is expanded and we are missing some cases when we create children variables. I cannot find a place where this is was actually used until now.

When you say copy value, VSCode uses that `evaluateName` and re-evaluates that as an expression instead of just using the value it was already provided. I did not expect it would take this **expensive** route, but there may be good reasons for it. This is where the failure happens.

```C
struct _myStruct {
   int x, y;
};
struct _myStruct myStructVar;
```

if I have a watch variable that looks like `&myStructVar` (you see the address of `myStructVar`) and I expand that you will see `x` and `y`. If you copy the value of x, the evaluate name is `&myStructVar.var` that is clearly wrong. That would give the address of `x` and not the value of `x` and in some cases (like in this case) may even be an invalid expression. We also had issues with anonymous unions which were a super pain to deal with.

I don't really know how many ways gdb provides results of variables back and the exact format other than by experimentation. I don't remember seeing a spec. we could follow.

I also don't have a C++ test case (or a way to compile) for my board. Bottom line is that there is a lot that can go wrong here. Also need to think of Rust expressions and variables. One simple thing I can do for my simple cases is to wrap all watch variables (non-simple ones) in parens and see what happens.

I will take a look at it when I have some time. This is going to be a bit involved and I have to spend sever consecutive days on it. Sorry, no quick fix. At least I found the problem.