Marus/cortex-debug #419: Specifying an RTOS option in launch.json on Mac for J-Link GDB Server results in wrong plugin path.

Archer36, 2021-03-23:
Hello, I just started experimenting with cortex-debug and still very new to embedded electronics, so my apologies if this isn't elegant.

I'm debugging a Particle Boron device using a J-Link probe using Particle Workbench. 

I setup my launch.json to look like:
`{
            "name": "J-Link Debugger",
            "cwd": "${workspaceRoot}",
            "executable": "${command:particle.getDebuggerExecutable}",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "nRF52840_xxAA",
            "interface": "swd",
            "serialNumber": "",
            "svdFile": "${command:particle.getDebuggerSVDFile}",
            "armToolchainPath": "${command:particle.getDebuggerCompilerDir}",
            "rtos": "FreeRTOS"
 }`

However, when the path to the FreeRTOS plugin as specified in https://github.com/Marus/cortex-debug/blob/c9532629d9abcf919020d5a765cbcf076d0df6ec/src/frontend/configprovider.ts#L172 is passed to the J-Link GDB Server, its taken as a relative path to the executable. This results in the J-Link GDB Server not being able to locate the RTOS plugins. This appears very similar to this issue: https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/146

To test I updated extension.js on my computer and updated the path to: 
`t.rtos="/Applications/SEGGER/JLink/GDBServer/RTOSPlugin_"`

This resulted in J-Link GDB Server finding the plugins to RTOS and working as expected. 

Let me know if I can provide any more info! Thank you for all the work on this great project!

FabianoGK, 2021-03-24:
Have you tried changing `cwd` in _launch.json_ to `/Applications/SEGGER/JLink` ?

haneefdm, 2021-03-24:
It is best to supply the full pathname to the `rtos` option in `launch.json` including the extension like .dylib or .dll. No guesswork, case sensitivity, etc. involved.

If you want relative pathnames, you will have to supply the extension. From SEGGERs documentation, see below

[image]

And no, please do not change `cwd`. That should be relative to where the elf file is, and where gdb is launched.

Then, there are other requirements to be met when using FreeRTOS especially. See

https://forum.segger.com/index.php/Thread/5595-SOLVED-SEGGER-FreeRTOS-plugin-with-FreeRTOS-v10-0-0-no-thread-found/

haneefdm, 2021-03-24:
Generally, most people run into issues with `uxTopUsedPriority` that is no longer provided/needed by latest FreeRTOS, but most GDB servers (JLink, OpenOCD) still expect it.

You have to define a global variable and make sure it survives compiler/linker optimizations. See

https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/33

If you google `uxTopUsedPriority` you will see how many people run into this problem

Archer36, 2021-03-24:
Thanks all. I wasn't aware that rtos would accept a full path. Just tried it out, the relevant entry in launch.json looks like

> {
>             "name": "J-Link Debugger",
>             "cwd": "${workspaceRoot}",
>             "executable": "${command:particle.getDebuggerExecutable}",
>             "request": "attach",
>             "type": "cortex-debug",
>             "servertype": "jlink",
>             "device": "nRF52840_xxAA",
>             "interface": "swd",
>             "serialNumber": "",
>             "svdFile": "${command:particle.getDebuggerSVDFile}",
>             "armToolchainPath": "${command:particle.getDebuggerCompilerDir}",
>             "rtos": "/Applications/SEGGER/JLink/GDBServer/RTOSPlugin_FreeRTOS"
>         }

However, I get the message:
`The following RTOS values are supported by J-Link: FreeRTOS or embOS. A custom plugin can be used by supplying a complete path to a J-Link GDB Server Plugin.`

When I do specify the extension, .dylib in my case, it appears to launch correctly! So the working config is:

> {
>             "name": "J-Link Debugger",
>             "cwd": "${workspaceRoot}",
>             "executable": "${command:particle.getDebuggerExecutable}",
>             "request": "attach",
>             "type": "cortex-debug",
>             "servertype": "jlink",
>             "device": "nRF52840_xxAA",
>             "interface": "swd",
>             "serialNumber": "",
>             "svdFile": "${command:particle.getDebuggerSVDFile}",
>             "armToolchainPath": "${command:particle.getDebuggerCompilerDir}",
>             "rtos": "/Applications/SEGGER/JLink/GDBServer/RTOSPlugin_FreeRTOS.dylib"
>  }

FabianoGK, 2021-03-24:
I apologize for the wrong suggestion. I didn't realize hovering over the `rtos` property does give a nice description of the options:

> _RTOS being used. For JLink this can be FreeRTOS, embOS or **the path to a custom JLink RTOS Plugin library**. For OpenOCD this can be eCos, ThreadX, FreeRTOS, ChibiOS, embKernel, mqx, or uCOS-III._

closed 2021-07-10 by haneefdm