Marus/cortex-debug #420: Semihosting, sending console commands

timoscs, 2021-04-01:
I'm trying to use semihosting with OpenOCD and Cortex-Debug. I've got almost everything working, but I can't seem to be able to send commands via semihosting. Basically I have a command line interface to the embedded system via semihosting, where I can do things like command IOs on/off etc etc, which is a big help in debugging.

What happens when I start my debug session is that the code compiles, is flashed (via OpenOCD with STLink), and the debug session starts, halted at the entry point. When I then continue, I see the semihosting output in the "Output/Adapter Output", which holds the as expected OpenOCD console. However, if I try to write text to enter a command (which is how it works under Eclipse), I get "Cannot edit in read-only editor".

I've attached a screen cap to show what I mean, the last three lines are output from my console running in the embedded device: 
[image]

If I run OpenOCD and arm-none-eabi-gdb separately in terminal, I can write to the OpenOCD console, so it seems that the Cortex-Debug extension needlessly makes the window read-only under VSCode.

Is there any way to use bidirectional semihosting in VSCode? Even a workaround, such as connecting to OpenOCD via telnet (or rather nc) will do (I've tried, but writing commands there also don't seem to go to the semihosting pipe), if this can't be done directly in VSCode. Or sending text to semihosting via the GDB console, which does allow writing. However, running gdb and OpenOCD completely separately is not a solution, since then I don't get any of the other debugging features in VSCode.

I'm on OS X Mojave and VSCode, Cortex-Debug and OpenOCD are the newest versions at the time of writing.

My launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Device Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": [
                "system/device_openocd.cfg"
            ],
            "overrideLaunchCommands": [
                "load ${command:cmake.launchTargetDirectory}/Device_crc.srec"],
            "preLaunchCommands": [
                "set mem inaccessible-by-default off",
                "monitor arm semihosting enable",
                "monitor reset halt"
            ],
            "interface": "swd",
            "internalConsoleOptions": "openOnSessionStart",
            "swoConfig": { 
                "enabled": true,
                "cpuFrequency": 100000000,
                "swoFrequency": 2000000
            } 
        }
    ]
}```

timoscs, 2021-05-06:
Anything on this? Please let me know if my question wasn't clear. 

Since this is a make-or-break for my transitioning to VSCode from Eclipse MCU, I'd be happy to spend a bit of time trying to implement this feature myself, if that's what  it takes? I've been browsing the cortex-debug source code, and I think I have a faint idea where to start, but I could certainly use some pointers on how to do it properly, and make sure that the feature is not already there and I'm just somehow missing it?

haneefdm, 2021-09-05:
Can you try the latest release? This feature was included in Version 0.4.0. Just enable semihosting like you have tried to do. No need for `internalConsoleOptions`, a console is now always running for the `gdb-server` (check the Terminal tab)

haneefdm, 2021-09-05:
Please let us know if you have issues. This has not been tested as well as the RTT terminals

closed 2021-09-05 by haneefdm

timoscs, 2021-12-23:
@haneefdm sorry it took quite a while to try this out, I was involved in other projects.

Anyway, based on a very quick test, it seems to be working perfectly! Thank you very much, this makes it possible to move our embedded dev to VSCode 👍 I'll post new issues if there's any kinks to work out, but this far no complaints.