Marus/cortex-debug #421: Zephyr RTOS support

fusedFET, 2021-04-05:
Hello, 
recently Segger added support for the Zephyr RTOS into Ozone. (See [Link](https://github.com/zephyrproject-rtos/zephyr/issues/23520).

It comes as a .js extension inside of Ozone, similar to how FreeRTOS seem to be handled. (There is a FreeRTOS.dll also, I guess the output of this DLL also works for Zepyhr, but is formatted differently using the .js file)

Is it possible to use this file to add Zephyr thread awareness support to Cortex-Debug?

haneefdm, 2021-04-13:
I really doubt that FreeRTOS.dll would work for Zephyr.

Just subscribe/follow that thread you posted and see if a Zephyr plugin for J-Link is available. Ozone may have support for Zephyr but if it does not use the same plugin mechanism, then it is not usable. Also, contact SEGGER

Cortex-Debug does not distinguish between RTOSes. That is between GDB and the specific gdb-server (OpenOCD, pyOCD, [J-Link GDB Server](https://www.segger.com/products/debug-probes/j-link/tools/j-link-gdb-server/about-j-link-gdb-server/), etc.

closed 2021-04-13 by haneefdm

fusedFET, 2021-04-29:
Reopening due to Segger adding the dll and the support of Zepyhr in the J-Link GDB Server with version 7.20.

haneefdm, 2021-04-29:
The support for Zephyr as a first-class RTOS was merged yesterday #428

You can always use any RTOS dll as long as it conforms to the JLink API. Btw, other than passing the RTOS plugin to JLing gdb-server, we don't much to do with it. No other magic.