Marus/cortex-debug #422: No source file name <src\main.rs> (from break-insert)

ashthespy, 2021-04-14:
Trying to get the debugger working with a Rust project and seem to run into this.

```
ERR No source file named e:\Projects\RustyPill\src\main.rs. (from break-insert "e:\\Projects\\RustyPill\\src\\main.rs:40"):
Error: No source file named e:\Projects\RustyPill\src\main.rs. (from break-insert "e:\\Projects\\RustyPill\\src\\main.rs:40")
```

My `launch.json` 
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            /* Configuration for the STM32F103C8T6 *Black Pill* board */
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build",
            "runToMain": true,
            "executable": "./target/thumbv7m-none-eabi/release/rusty_pill",
            "device": "STM32F103C8",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f1x.cfg"
            ],
            "postLaunchCommands": [
                "monitor arm semihosting enable",
            ],
            "svdFile": "${workspaceRoot}/.vscode/STM32F103.svd",
            "showDevDebugOutput": true,
        }
    ]
}
```

FWIW, `main.rs` does exist ;-)

```
.
├── Cargo.lock 
├── Cargo.toml 
├── memory.x   
├── openocd.cfg
├── openocd.gpb
├── src        
│   └── main.rs
└── target
    ├── CACHEDIR.TAG
    ├── debug
    ├── release
    └── thumbv7m-none-eabi
```
I'm on Windows if that makes a difference.

haneefdm, 2021-04-14:
GDB/Cortex-Debug is using the full path-name `e:\Projects\RustyPill\src\main.rs` does that exist?

Is it possible to share your executable? You can trim your source to just the main.rs and create that executable if you like.

ashthespy, 2021-04-14:
> GDB/Cortex-Debug is using the full path-name `e:\Projects\RustyPill\src\main.rs` does that exist?

The path exists, but it looks like it is a `cygwin` based path and not a windows path.
``` > Get-Location      
Path
----
E:\Projects\RustyPill
```
Looking the debugger console, I also see
```
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from E:\\Projects\\RustyPill\\target\\thumbv7m-none-eabi\\release\\rusty_pill...\n"}]}

```
Which seems to have the correct path notation. 

>Is it possible to share your executable?

[rusty_pill.zip](https://github.com/Marus/cortex-debug/files/6312499/rusty_pill.zip)

Thanks for taking a look! :-) 

I also noticed that gbd launched by the debugger isn't picking up my correct `$HOME` path for the `add-auto-load-safe-path` .  I've set up the right variables and it works fine from a terminal session..

haneefdm, 2021-04-14:
I looked inside the executable and I see symbols but no references to any source files. I don't know how you compile your program but debug information is missing. Zero file names. So, there is no way to set breakpoints (gdb or Cortex-Debug or VSCode) using source file/line references. You can find this out yourself using `objdump`

```sh
arm-none-eabi-objdump --syms -C rusty_pill
```

I never compiled a rust program but in the end, I know we should get a valid elf file and it should have the proper debug info. With a C program for instance you have use one of `-g -g1 -g3` on the compiler and linker

ashthespy, 2021-04-14:
Ah! I have much to learn with embedded Rust. I was of course compiling it with no debug symbols. Thanks for taking the time to point me in the right direction! 

I'll check back in the morning and confirm..

haneefdm, 2021-04-14:
Closing for now, you can continue to report progress/issues

closed 2021-04-14 by haneefdm

ashthespy, 2021-04-15:
Happy to report that it was user error as usual - built it with debug symbols, and things are working as expected.