Marus/cortex-debug #425: Support for CMSIS Packs

itavero, 2021-04-15:
Previously I mainly used Eclipse CDT (with GNU MCU Eclipse). In this environment [CMSIS Packs](https://developer.arm.com/tools-and-software/embedded/arm-development-studio/learn/docs/ds-5-migration/cmsis-packs) were used to get information about the available registers and such of a microcontroller.

Is this also supported by this Visual Studio Code extension?
If not, would it be possible to add support for it?

I read something about SVD files, which I think serve the same purpose.
Unfortunately, I can't seem to find SVD files for any of the recent microcontrollers (i.e. NXP LPC55xx series) we've been using.

haneefdm, 2021-04-15:
SVD is part of the CMSIS standard. I doubt Eclipse goes through CMSIS source files to figure out the registers' info. The SVD files mostly provide the device's peripherals. That info does not exist in CMSIS source packages

Please contact NXP for SVD files. See thread below.

https://community.nxp.com/t5/LPC-Microcontrollers/svd-for-LPC55S69/m-p/859098

They seem to be available only upon request?!?! I don't know.

haneefdm, 2021-04-15:
Another way is to create a project in Eclipse and see what is in the SVD tab for Debug/Run. It probably has the pathname to the SVD file.

closed 2021-04-15 by haneefdm

itavero, 2021-04-15:
GNU MCU Eclipse downloads the CMSIS Packs directly from KEIL apparently: https://eclipse-embed-cdt.github.io/plugins/packs-manager/

itavero, 2021-04-15:
@haneefdm May I ask why you closed this feature request?

haneefdm, 2021-04-15:
VSCode is not a project manager and if NXP (or someone else) can provide a
plugin to do the same.

Cortex-Debug is just a debugger and does not aim to be a project manager.
Besides as you can see in the link provided, it is still experimental, ST
hasn't joined it. We cannot validate the authenticity of anything there.

You can always re-open an Issue. When I feel like there is nothing I can
add or I see things beyond our charter, I generally close it.

BTW, if the pack contains SVD files from all the vendors, then we are
talking GB's of stuff and I am not sure what their update mechanism is.

itavero, 2021-04-15:
I don't think you get what a pack actually is. It's a file per processor with info like the registers and such.

The Eclipse plugin doesn't download them all. It just provides a convenient interface to select the processors you use and download the info (pack) for those, instead of having to search tirelessly for a file online.

I still feel that adding something like this to the Cortex-Debug plugin would improve the user experience a lot.

RisinT96, 2021-04-15:
AFAIK the CMSIS pack is just an archive with a bunch of stuff in it, if you open it with 7z or some other archive tool, you can find the SVD inside.

itavero, 2021-04-15:
@RisinT96 I wasn't able to find the specification in between my meetings today. If this is indeed the case, then adding support for this is probably rather straightforward. I've never written a VSC extension, but I'll see if a can add it and provide a PR, if I have some time.

RisinT96, 2021-04-15:
I might be mistaken though, or it might be non-standard, but I'm pretty sure I found svds inside SiLabs device support packs.

itavero, 2021-04-15:
Just realized I can make a separate plugin that extends the Cortex-Debug plugin, by using the `registerSVDFile` extension method.
There's no "unregister" method, right?

haneefdm, 2021-04-15:
@itavero I do get what CMSIS packs are. Authored them, debugged them, etc. A CMSIS pack can be many things, a board/device support package, middleware, example projects, etc. So, it goes much beyond SVD files. Board/Device packs can run into 100's of MBytes just to see what is in them -- and an SVD file may be buried in there somewhere and there are multiple versions of each pack.

See this plugin maintainer and issues s/he is having and see how the future of that plugin

https://eclipse-embed-cdt.github.io/plugins/packs-manager/
https://eclipse-embed-cdt.github.io/plugins/packs-manager/#future-developments

If someone wants to take on creating/maintaining a VSCode plugin, why not. We (speaking for a grand total of two part-time contributors) don't have the bandwidth for that and I don't want to become customer support for all the vendors' issues especially those related to versions and potentially multiple versions of the truth. Right now, we (mostly) don't even look at the device name. We just pass it on to a gdb-server that wants it.

I ask again, if you already have a trusted mechanism (maybe Eclipse or MDK) to download said packs, why not just use that. As I understand it, it just deposits stuff on disk and you can just point Cortex-Debug to a particular SVD file.

NB: We used to distribute SVD files ourselves in the past. Finally, we gave up too

haneefdm, 2021-04-15:
Yes, you can create an extension to use the `registerSVDFile`. You can see an example here...

https://github.com/Marus/cortex-debug-dp-stm32f1

You will probably need hundreds (if not thousands) of regexp's to support all the vendor's devices and still have to have the particular CMSIS pack/SVD on disk. This is what I meant by

> NB: We used to distribute SVD files ourselves in the past. Finally, we gave up too