Marus/cortex-debug #426: Fails to Connect to GDB Server

microwavesafe, 2021-04-15:
I was forced to upgrade my PC, I'm now running VSCode 1.55.2 on Pop_OS 20.04 with JLink 6.98. I've installed the latest cortex-debug extension, but it won't connect.
The Adapter output is
```
JLinkARM.dll V6.98e (DLL compiled Mar 29 2021 14:21:59)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F070C6
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F070C6
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Feb  4 2021 12:58:41
Hardware: V10.10
S/N: 600109529
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.31 V
Listening on TCP/IP port 50000
Connecting to target...
```
The debug console with showDevDebugOutput: true is
```
Reading symbols from '/home/Documents/lvs/build/bin/output.elf'
Finished reading symbols
Please check OUTPUT tab (Adapter Output) for output from JLinkGDBServer
Launching server: "JLinkGDBServer" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "STM32F070C6"
Launching GDB: "/opt/arm_gcc/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
1-gdb-set target-async on
```
Which looks likes everything is called correctly. The connection never succeeds and the debug buttons disappear. If I run JLinkGDBServer directly with the same parameters, it connects fine.
```
$ JLinkGDBServer -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F070C6
SEGGER J-Link GDB Server V6.98e Command Line Version

JLinkARM.dll V6.98e (DLL compiled Mar 29 2021 14:21:59)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F070C6
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F070C6
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Feb  4 2021 12:58:41
Hardware: V10.10
S/N: 600109529
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.31 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...^CRestoring target state and closing J-Link connection...
Shutting down...
```
There doesn't seem to be any error messages for me to look into, it just doesn't connect, even though the same command at the terminal works fine. Does anyone have any idea what's going on?

haneefdm, 2021-04-15:
In the Cortex Debug launched server, I don't see the following but it could have been buffered and not yet displayed
```
Connected to target
Waiting for GDB connection...
```
We wait for 10 seconds to see that last line before launching gdb which looks like we have done.

Can you launch gdb manually in another terminal while running the server like you did from the command line and issue the same commands Cortex-Debug has and see what happens?

microwavesafe, 2021-04-15:
That's it! I should have tried to manually connect to the GDB server myself. I ran this, after starting the JLinkGDBServer
```
$ /opt/arm_gcc/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb
/opt/arm_gcc/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```
An error I can fix
```
sudo apt-get install libncurses5
```
Now it runs perfectly! Thank you for the nudge in the right direction.

closed 2021-04-15 by microwavesafe

haneefdm, 2021-04-15:
Darn it, I always forget that gdb issue for Linux. It came up soooo many times and yet, I still forget to mention.