Marus/cortex-debug #430: Unable to see any variable values

ajmilford, 2021-04-28:
Using this plugin with JLINK.

Here is my launch.json

` "configurations": [
    {
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/BUILD/LPC1768/GCC_ARM/AirWatchFWBeta_application.elf",
        "name": "Debug Embedded",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "jlink",
        "serverpath": "C:/Program Files (x86)/SEGGER/JLink_V512f/JLinkGDBServerCL.exe",
        "device": "LPC1768",
        "interface": "swd",
        "runToMain": false,
        "preLaunchTask": "msbuild"
        
    },`

program loads to flash fine, breakpoints work, single stepping, step in, step out etc.
But I cannot view values of any variables.

In the globals variables list, if I hover the mouse over any variable the popup shows "<data variable, no debug info>"
Nothing shows in local
If I add a variable to the watch - none show a value.

This is an mbed OS project in case that makes a difference.

Have I got something wrong in config?
What can I do to fix this?

Thanks

haneefdm, 2021-04-28:
* Did you compile (and link) with full debug enabled (usually -g or -g3) assuming you used GCC. If another compiler, equivalent debug options
* Do you see a stack-trace (The Stack window)? I don't think JLink supports Mbed OS, so it will not show threads but it should still see it as a normal program with just one thread
* In the `Debug Console`, after you stop at a breakpoint or pause the program can you use gdb commands to print local/global variables? If this does not work, neither will Cortex-Debug

diggit, 2021-05-24:
Hi,
because I see similar issues, I'll try to provide more info.
My sources are C++ and compiled with debug symbols.
- Archlinux
- arm-none-eabi-gcc 11.1
- arm-none-eabi-gdb 10.2
- vscodium 1.56.2

My  "Variables" window is pretty much empty, especially statics, but globals too. This is long term state for me.
[image]

`info variables` gives much more rich output (attached as file to keep this thread scrolling friendly :)
[gdb_variables_info_output.txt](https://github.com/Marus/cortex-debug/files/6533792/gdb_variables_info_output.txt)

Stack trace sometimes disappears. Dirty workaround is disconnecting and reconnecting to target in "attach" mode. If It occurs again, I'll provide stack trace obtained from gdb for comparison and give more details.

haneefdm, 2021-05-24:
@diggit What is displayed in the `VARIABLES` window is not the same as `info variables`. The latter prints all global static variables in your program. The Variables window displays only what is currently visible for a given stack frame.... which are 'Locals', 'all Globals' and 'current file statics'

Try `info locals` to actually get local variables and they should match... if the stack frame matches.

On why the stack frame is empty, that is a different question. If there is no stack, the Local/Statics will also be empty as there is no reference point. This has been happening with a recent version of VSC and I haven't been able to reproduce it.

diggit, 2021-05-24:
> The Variables window displays only what is currently visible for a given stack frame.... which are 'Locals', 'all Globals' and 'current file statics'

Oh, that is quite important information!
IMO, it would be nice to have this behavior configurable. I'd prefer extensive list with fulltext search.

About glogals... you can see global object `fd1` on screenshot below.
[image]

it's not listed in globals view. Is there some other rule?

It's also listed in "artificial" file. Could that be a problem?
```
File
<artificial>:
...
57:	xhal::stm32::canfd::StaticDevice<4096, 4096> fd1;
...

haneefdm, 2021-05-24:
@diggit Could you run the following command and attach the results. The output may be super large

```sh
arm-none-eabi-objdump --syms -Wi -C <elf-file-name> > objdump.txt
```

This is how we determine the statics and globals. If a symbol is not marked global, then we would not know about it. But, I want to see what `fd1` IS marked as.

diggit, 2021-05-24:
Sure here it is:
[objdump.zip](https://github.com/Marus/cortex-debug/files/6534856/objdump.zip)
and here is elf, in case you need some more info
[binary.zip](https://github.com/Marus/cortex-debug/files/6534863/binary.zip)

haneefdm, 2021-05-24:
`fd1` is marked as a local variable with no file association. There will be hundreds (or even thousands) of such things sometimes due to compiler/linker magic.

```
20001b00 l     O .bss	00003160 fd1
```
The `l` after the address says it has local scope (generally file scope), but it is not even associated with a file.

I don't know how gdb determined what it did

diggit, 2021-05-25:
`fd1` is used in only one file. Maybe that's the reason.

About that missing backtrace, I mentioned earlier...
[image]

Then disconnected and attached back
[image]
(I am not sure, why it stepped into span operator[], even thought, I did not request stepping )

haneefdm, 2021-05-26:
When you detach, a continue is issued. That is the gdb spec. Some gdb-servers do that and compliant ones do that and some just leave it paused. Think of debugging a unix/linux daemon. you don/t want to leave it paused even by mistake.

haneefdm, 2021-05-26:
Yup, the stack trace thing is a big one. Wish I could duplicate it. If it is repeatable, are there steps I can follow to help duplicate the issue.

haneefdm, 2021-09-05:
We have no further feedback for a while. Closing issue

closed 2021-09-05 by haneefdm

diggit, 2021-09-08:
Well, I was not able to replicate this issue. It just happens from time to time. Issue still exists. At least the backtrace absence.