Marus/cortex-debug #431: Support for Cortex-A series

r06922019, 2021-04-29:
Hi,
I'd like to know whether this plugin supports the same feature as Cortex-M series
Thanks.

haneefdm, 2021-04-29:
Could you please expand on your question? The title says Cortex-A, the description says Cortex-M

As far as Cortex-M series, we rely on GCC/GDB and gdb-servers like (openocd, jlinkgdbserver, etc). We are 32-bit, little-endian only. You can also use other compilers (IAR, ARM, etc.) so long as they produce a valid ELF executable debuggable by GDB.

r06922019, 2021-04-29:
Hi @haneefdm,
Sorry I didn't make it crystal clear.
I've tried this on Cortex-M platforms and worked really well.
But when I try it on a Cortex-A chip, the plugin seems to fail to translate the current pc to the corresponding position in the source code.
I was wondering and would like to confirm if this plugin supports Cortex-A. 
If so, I must have set something wrong.
Thanks.

haneefdm, 2021-04-29:
Other than Disassembly, we don't really care about the PC. It is gdb. In `Debug Console` you can issue gdb commands and see what it reports. Print the stack trace, etc. It may not also work if you compiled/linked with no debug information (use -g or -g3). Maybe the memory window may have an issue with addresses

Of course, I am assuming a Cortex-A32 (32-bit processor). Anything with 64-bit addresses will not likely work.

haneefdm, 2021-07-10:
No response in a while. Cleaning up issue list

closed 2021-07-10 by haneefdm