Marus/cortex-debug #432: SWO Output on a Cortex-M4 with Cortex-M3 TPIU

gtaska, 2021-04-30:
I have been trying to get SWO output working on a Cypress PSoC 6 device's Cortex-M4 core.

With a (presumably valid) swoConfig block in launch.json, I get the following error message:

> Failed to launch GDB: Cannot access memory at address 0xe0040304 (from interpreter-exec console "BaseSWOSetup 49")

After [a lot of sleuthing](https://community.cypress.com/t5/PSoC-6-MCU/PSoC6-Cortex-M4-SWD-SWO-debug-output/td-p/275313), I have been able to determine **that the error is valid** - because the PSoC6's Cortex-M4 hardware design uses a Cortex-M3 TIPU block (which is located at 0xe008e000, rather than 0xe0040300).

_Any thoughts on whether it is possible to get/force cortex-debug to use a Cortex-M3 TPIU with an otherwise Cortex-M4 core?_

haneefdm, 2021-04-30:
Unfortunately, there is no good way for Cortex-Debug. Technically, Cortex-Debug should do a discovery to see the actual base addresses are for a given debug component. However, since you know the correct base addresses, you can modify the `gdbsupport.init` in the extension install-dir. Note that when a new version of the extension is installed, your changes will go away.

Mine is here `~/.vscode/extensions/marus25.cortex-debug-0.3.12/support/gdbsupport.init`.

Depending on the PSoC6 device, it still might not work because of clocking and flushing issues. Let me know how things work out

gtaska, 2021-04-30:
Yep, thanks. Already playing around with that concept.

It doesn't crash when I attach anymore, but no output on the SWO pin yet (but that may be due to other issues, still to be uncovered).

At least I know I'm moving in the correct direction.

closed 2021-07-09 by haneefdm