Marus/cortex-debug #433: Always make an error in the clock setting function

yueduz, 2021-05-03:
I'm developing stm32h750. The same code uses stm32cubeide and debugs normally. When using Cortex Debug to debug, he made an error during the clock setting process. The location of the error is not fixed, but they are all in the clock setting function, and occasionally can run normally. I noticed that downloading using Cortex Debug is slower than stm32cubeide. Did Cortex Debug delay the time when reading the svd file after downloading, causing an error?

haneefdm, 2021-05-03:
As an experiment, you can comment out the SVD file in launch.json to see if it improves.

Cortex-Debug does not download the program to the device. that is done by gdb and the gdb-server (link). If JLink used a slow interface clock, then it would be slower. You can check the 'Output' Window and select 'Debug Adapter'. I am sorry, I don't how that translates to your language

yueduz, 2021-05-03:
Now I don't use svd files, and reducing the frequency doesn't work anymore.

yueduz, 2021-05-08:
[image]
I want to change Single run mode for a try, but how do I configure Single run mode in cortex-debug?

haneefdm, 2021-05-08:
In `launch.json`, try `"serverArgs"` you can pass any command-line argument you want.

In this case, we request the server to exit if we started it. Is that not happening? It doesn't matter what single-run mode is, we use the same policy across all gdb servers. JLink is just one of them.

yueduz, 2021-05-09:
I don't quite understand what you mean, I use Google Translate to translate our conversation. I used the "serverArgs" parameter and did not add it to the start command of jlink gdbserver, it seems to be added to the gdb client.

On the other hand, I manually run JLinkGDBServerExe. After a failure, I selected the option of image marking, and he succeeded. How can I add this option to cortex-debug?
[image]

yueduz, 2021-05-09:
> I don't quite understand what you mean, I use Google Translate to translate our conversation. I used the "serverArgs" parameter and did not add it to the start command of jlink gdbserver, it seems [...]

I was wrong, the parameters were passed in, but an error was reported in the clock function

yueduz, 2021-05-09:
[image]
I found another difference, cortex-debug starts "jlinkGdbServer" Target connection timeout is always 0 ms

haneefdm, 2021-05-09:
If you want a timeout, then add `"-timeout", "5000"` to your `"serverArgs"`.

That is a loooong time btw. I can't help you with your clock function. There are too many things that can be going on and none of them are probably related to Cortex-Debug.

Since there are many command-line options to the Jlink gdb-server ... see Section 4.5 below

https://www.segger.com/downloads/jlink/UM08001

yueduz, 2021-06-16:
I'm very sorry that I opened this question again. Because I think I found the real reason. After carefully comparing the output of stm32cubeIDE, I found that commenting out lines 42 and 45 of jlink.ts, I was able to get my board to work normally.
[image]

yueduz, 2021-06-16:
[image]
I use the overrideLaunchCommands configuration item, but he pops up an error as shown in the figure

yueduz, 2021-06-16:
```
            "overrideLaunchCommands": [
                "monitor reset",
                "-target-download",
                "-enable-pretty-printing"
            ]
```
I finally found a way to configure

closed 2021-06-16 by yueduz