Marus/cortex-debug #434: Mac OS not hitting breakpoints with ST-Link

samchamberlin, 2021-05-05:
Using this plugin with ST-Link and breakpoints don't seem to be working on macOS Big Sur. I compiled with full debug enabled. In the debug console I get this output from the "info breakpoints" command: 
[image]

Has anyone run into this problem? Any help is appreciated!

haneefdm, 2021-05-05:
Can you make it work using the command-line tools? Breakpoints, while we set them from the extension, we just wait until gdb (which waits on st-util) to report a hit. If that doesn't work, neither will Cortex-Debug

samchamberlin, 2021-05-06:
Thanks @haneefdm, here's the debug console with `showDevDebugOutput` set to true:
[image]

It seems to run just fine - the STMicrocontroller shoots serial output out to me correctly. I'm not familiar with setting breakpoints using command-line tools if that's what you mean. 

Thanks for the help!

ubidefeo, 2021-06-08:
@samchamberlin 
do you happen to have your breakpoint right before the end of a function or at the exit point of a code block?
I have noticed that if the breakpoint is placed as follows

```c++
if(condition){
   counter++;
   secondCounter--;
   thirdCounter += counter; // place breakpoint here
}
```

it won't be caught.
same happens when you have two consecutive breakpoints and on the first one you do `step into`: when doing `continue` it won't stop at the second breakpoint

closed 2021-10-06 by haneefdm