Marus/cortex-debug #436: Unable to pass while(!Serial) in debug

simon88, 2021-05-18:
Hi,
I try to use cortex debug in Vscode on my mac with a personal board, build with an ATSAMD21E18A, I've configured laucnh.json, here is the begin of my setup function:

setup(){
  SerialUSB.begin(115200);
  while(!SerialUSB);

  Wire.begin();  
  Wire.setClock(400000UL); 
  ....
}

If I put a breakpoint at the first line SerialUSB.begin(115200); everything works, but the debugger never pass second line while(!SerialUSB); I never reached  Wire.begin();  when I debug, any ideas? Need I configure something in laucnh?json to use SerialUSB maybe ?
My Board is connected to my MAC with an USB cable and I use an Atmal ice for debugging with openocd.

Thanks

haneefdm, 2021-05-18:
Hi @simon88 , Cortex-Debug is a front-end to `gdb`. It does not know what device/board/connections you are using or know anything about your firmware (other than the `exe` name and symbols within it). When gdb reports something, we report the same.

If you can gdb to work from the command line, then you can more than likely make it work with Cortex-Debug.

Go ahead and post your `launch.json` but I doubt there is anything in there.

haneefdm, 2021-07-10:
No response in a while

closed 2021-07-10 by haneefdm