Marus/cortex-debug #438: Default Target Hostname

nosajthenitram, 2021-05-20:
File: jlink.ts 

'''
public initCommands(): string[] {
        const gdbport = this.ports[createPortName(this.args.targetProcessor)];

        return [
            `target-select extended-remote localhost:${gdbport}`
        ];
    }
'''
There's an issue when using cortex-debug from WSL2. The WSL2 session does not properly resolve 'localhost' and is unable to connect to the launched instance of jlink gdbserver. Can you please add gdbtarget as a valid option for the current interfaces that have a default hostname of 'localhost'.

haneefdm, 2021-05-20:
We have discussed this and there are lots of complications with changing that. And, you can't just do that for JLink. Please see issue #402

https://github.com/Marus/cortex-debug/issues/402#issuecomment-785864841

nosajthenitram, 2021-05-20:
Thanks for pointing me to the open discussion.

closed 2021-05-21 by nosajthenitram