Marus/cortex-debug #440: Cortex registers view shows decimal instead Hex

Delsian, 2021-06-03:
During debug "Cortex peripherals" shows numbers in HEX, but "Cortex registers" in Decimal.
I see no sense to show decimals for registers, probably better to set display format to Hex

[image](https://user-images.githubusercontent.com/2592414/120633601-00e19900-c473-11eb-9fb2-86624ece0289.png)

lagerholm, 2021-06-04:
See #364.

Delsian, 2021-06-04:
@lagerholm Thank you, right now it was resolved by adding
```
            "postAttachCommands": [
                "set output-radix 16"
            ],
```
into launch.json file. But it's a tricky workaround, I prefer to see this in the base installation.

haneefdm, 2021-08-16:
See most recent release V0.4.2. You have support from the GUI to do this, is persistent and switchable at runtime. You can also do it as a User setting and override per workspace.

closed 2021-08-16 by haneefdm