Marus/cortex-debug #442: Is an hotplug debug possible?

ziki77, 2021-06-10:
Is it possible to do a hotplug connection without flash and reset of the uC?

haneefdm, 2021-06-10:
If I understand you correctly, you want to attach to a running program in a uC. In your `launch.json` you have to
```JSON
    "request": "attach",
```
No programming or reset will be done. but there will be a `halt` will be issued. Please close this issue if that is what you wanted

ziki77, 2021-06-11:
Hi,
it is exactly what I need and it works! Thanks
Is there the possibility to avoid the halt at attach?

haneefdm, 2021-06-11:
Yes, you can override what happens on an attach. In launch.json, start typing `"over` and you will see a bunch of customizations. But you have to know what you are doing. If no halt is done, breakpoints cannot be set. I don't know what else will break. You are better off using a `postAttach...` command to continue but I can't guarantee what behavior you get.

I can't remember exact spellings, I just rely on VSCode Intellisense to get command names and help.

closed 2021-06-11 by haneefdm