Marus/cortex-debug #445: Run Without Debugging

pearson, 2021-06-11:
Should the VS Code "Run -> Run Without Debugging (CTRL-F5)" work correctly with cortex-debug? What I am finding is that the debugger is activated and the program halted at main, just like it does with "Start Debugging".

It could be that I'm missing a needed configuration, but I'm at a loss for finding documentation on flashing an MCU with J-Link through VS Code that does not involve debugging. Any thoughts on this would be appreciated!

haneefdm, 2021-06-11:
It is not supported. We do the same thing for both. What should it mean for run without debugging in an embedded environment? Isn't it the same as hitting the reset button? And, what should the gdb/gdb-server do? Answers were not clear to us. I know Eclipse has such functionality but not sure what it does and how useful it is.

Did you check the Output Tab (in the Debug Adapter section). All output from the gdb-server(s) go to that area and you may see plenty of chatter.

pearson, 2021-06-11:
> Isn't it the same as hitting the reset button?

I suppose it is, in a way, except that it removes the extra step of having to physically press the reset button as well as avoiding a context switch in VS Code. PlatformIO also gives a separate "Upload" button that flashes the device and resets it, which is very convenient for making quick changes that don't require the debugger. It also keeps VS Code's side bar set to "Explorer" rather than switching it to "Run and Debug".

A couple of use cases that come to mind are:
- when a bunch of quick changes are being made to the way objects on a display are laid out
- when the reset button isn't easily accessible
- when testing power consumption modes with release code. I'd like it to upload and reset in one action so I can get to monitoring power use.
   - that Mbed example in the other issue thread is a good example, since `ThisThread::sleep_for()` will disconnect the debugger anyway if sleep modes are active. 

Perhaps it's simply wanting to have the IDE do as much for me as possible (avoiding that reset button press), and to stay in the same state so I can quickly open files as needed without having to click "Explorer" first. Laziness, perhaps. :smile:

I had thought that changing "runToMain" to false might start the program without any halts, but VS Code gives me this red bar "Exception has occurred" message. I assume it's related to the `Not implemented stop reason (assuming exception): undefined
` message in the debug console, although that is exactly the same message that appears when "runToMain" is set to true.

[image]

haneefdm, 2021-06-11:
When you set runToMain to false and a launch is done, the default behavior is to do a 'reset halt' to get the PC and stack into a known state. Well, the halt is the issue. So, what you would like is to do a reset but no halt and ignore all breakpoints including the 'runToMain'. I am not sure but even just doing a 'reset' will halt at the entry point (reset vector) You can experiment with this as well using the `override...` launch options

Did you try experimenting with `runToEntryPoint`? That will be the new `runToMain`. There are other things that occur at the very beginning (reset). OpenOCD/JLink for instance will try to determine if the program uses an RTOS and change its behavior. 

Btw, I never use `runToMain` because I am never interested in it and I don't want to waste a breakpoint. This is why the default is `false`

haneefdm, 2021-06-11:
We also have to figure out what the debugger should do when launch type is `attach` to a running program.

pearson, 2021-06-14:
I had not tried `runToEntryPoint`. I see there's a mention of it in some patch notes, but I don't think it's mentioned in the documentation so it's not clear how to configure that  entry point name. (At least it's not documented in the wiki here.) 

I picked up that `runToMain` from some example somewhere, I think. That's good to know, though.

I'm curious about the `attach` question. Is there something special you're thinking of? It seems that, in other IDEs, the debugger will connect to the MCU and leave it running, either waiting for a pause command or to hit a breakpoint.

One question I do have for you: how are you doing the initial flash to a J-Link device? Is it through GDB? I'm thinking of making a simple task to flash the device, and I've experimented with the command-line GDB, but I haven't seen the same J-Link "flashing" dialog box that appears when starting a debug session with cortex-debug.

haneefdm, 2021-06-14:
`"runToEntryPoint": "main"` is the same as `"runToMain": true` what is expected is the name of a function for a value. Empty string means don't do anything.

About "attach", I have seen it both ways. If you have breakpoints for instance, there is no way to set them (via gdb) if the program is not halted. The author of the extension decided to halt at attach and I agree with that. This is the expected/recommended behavior of GDB itself and what I am used to debugging daemons in a Unix'ish environment.

https://sourceware.org/gdb/onlinedocs/gdb/Attach.html

> The first thing GDB does after arranging to debug the specified process is to stop it. You can examine and modify an attached process with all the GDB commands that are ordinarily available when you start processes with run. You can insert breakpoints; you can step and continue; you can modify storage. If you would rather the process continue running, you may use the continue command after attaching GDB to the process.

That is the behavior we try to emulate.

Yes, more often than not I let gdb do the programming. But there are times I issue commands directly to the gdb-server -- because the executable (elf fiole) does not have all the information needed to program the device properly. So, it depends.

pearson, 2021-06-15:
Is the output from the programming available? When the J-Link flash window only pops up briefly, it would be nice to see if anything was actually flashed. The use case I ran into is where I thought a parameter was making a compiler change, but in reality the binary was identical. Flashing manually gave this message which helped clarify the problem: `J-Link: Flash download: Bank 0 @ 0x00000000: Skipped. Contents already match`

Would you mind pointing me to the code in the extension that takes care of the flashing? I wouldn't mind looking at it and perhaps working on the flashing mode I'm interested in.

pearson, 2021-06-15:
Just to clarify that... I don't see a flash message in my J-Link adapter output.

closed 2021-06-15 by pearson

haneefdm, 2021-06-15:
There is no code per se to flash the device. It is a gdb-command that is issued `target-download` which is part of the launch sequence that you can override. Just search for 'target-download'

pearson, 2021-06-16:
Another question for you, if you don't mind...  

Is there a parameter to ensure the project is built before debugging, perhaps similar to the cmake-tools cmake.buildBeforeRun? Or would such a thing need to be put into the preLaunchCommands?

Thanks!

haneefdm, 2021-06-16:
Yes, this is part of VSCode itself. All VSCode debuggers have that. It is called "preLaunchTask" which can run a task defined in `tasks.json. Search here...

https://code.visualstudio.com/docs/editor/debugging

pearson, 2021-06-16:
Thanks, @haneefdm . I do appreciate the help climbing the steep entry curve into this.

pearson, 2021-06-18:
@haneefdm , do you know of any way to bring the SWO output window to the front when a debug session starts? I've been searching for information on this, but haven't found anything.

Thanks!

EmanuelW, 2022-03-24:
@pearson I think I accomplished what you where originally asking for by creating a separate launch task and add the following "postLaunchCommands"

```yaml
"postLaunchCommands": [
    "monitor reset run",
    "detach"
]
```

That does not make "Run Without Debugging" work ofc but does provide that functionality.

As for the discussion about use-case, I use this as a convenient way to flash a debug-fw to units that will run over night or just when I need to detach the probe for some other purpose.

pearson, 2022-03-24:
Thanks for the tip, @EmanuelW ! I'll give that a try.

haneefdm, 2022-03-24:
May I ask what the problem is with "Run Without Debugging"? It has been implemented now. This particular issue has a mishmash of various things now. A LOT of things have changed since mid last year. We should file a new issue with details (copy of launch.json, output of Debug Console, gdb-server, etc.). It should work with your regular launch configuration but having a separate one is not a bad idea.

pearson, 2022-03-24:
@haneefdm , to be honest, I haven't looked at those features in Cortex-Debug in a while. I just got a ping when @EmanuelW 's message was posted. 

Currently I have my own hotkey triggered upload/reset script which does the "run without debugging" job, but Cortex-Debug is still there for debugging, of course!

haneefdm, 2022-03-24:
No worries :-) Thanks for responding.

francisjeanneau, 2022-04-21:
> May I ask what the problem is with "Run Without Debugging"? It has been implemented now. This particular issue has a mishmash of various things now. A LOT of things have changed since mid last year. [...]

Hello there, could you refer us to any documentation regarding this feature (since you seem to say it has been implemented). I think what OP wanted (and I) only wanted to hit the run button to flash & reset the device without debugging (like you mentioned Eclipse does). What would be the cleanest solution as of today to do this?

haneefdm, 2022-04-21:
This is a standard VSCode feature (`Run Without Debugging`) and we tried to respect that. Not many people use it so if you run into issues, let us know.

[image]

VSCode is a minimalist and does not provide a button for it. But the command to do it exists and there is a keybinding to it.

francisjeanneau, 2022-04-21:
Oh I see, just tried CTRL-F5 and it seems to be working. However, VS Code seems to be keeping a "debugging instance" running nonetheless (see screenshot). This forces the user to stop it before reflashing. In other words, it seems like you can't hit CTRL-F5 multiple times without hitting the "stop" button from the debugging instance in between. Does cortex-debug have any control over that?

In any case the current behavior is helping, to me this is the last minor annoyance that prevents the feature from being perfect!

[image]

haneefdm, 2022-04-21:
Yes, that is the way it works. The pause button, I thought was disabled. But the buttons to restart/kill/reset will exist.

The main difference is breakpoints are not set so the program will not stop on its own. No other difference.

Btw, there is no way for us to run a program without using GDB and the GDB-Server. Everything still happens via those programs and the probe you are using. Remember we have embedded programs and not something that is running in the host OS.

Also, things like SWO/RTT still work even without debugging.

haneefdm, 2022-04-21:
> In other words, it seems like you can't hit CTRL-F5 multiple times without hitting the "stop" button from the debugging instance in between. Does cortex-debug have any control over that?

Never tried that so I don't know. Normally, VSCode does not allow the same configuration to run more than one at a time.

Try it and if it is launching multiple instances fine... and we will handle it as each instance is a totally separate instance of Cortex-Debug. What will happen though is contention at the USB connection and the second one won't start as USB is busy with the first one. Most gdb-servers will error out. Again, we never talk to any HW directly -- not the host USB or serial-port or the probe or your board.

francisjeanneau, 2022-04-21:
It tries to launch multiple instances, but fails since the debugger is already in use by the first instance. I think the question is "why does the debugger stay attached when doing `Run Without Debugging`?".

As someone coming from Eclipse/SW4STM32, I am used to pressing "Run" and have the IDE compile, flash & reset device. Nothing more, nothing less. Nothing stays attached to the debugger. The debugger, after flashing the device, is freed. I am not well versed in the underlying gdb server but my guess is that it detaches after flashing & resetting the device. The advantage of that feature of Eclipse is that you minimize the steps needed to _only_ flash the device, a single click and it's done. You can redo it without any other intermediate step. To mirror this with VS Code & cortex-debug, I would have to be able to:

1. change code, save file
2. CTRL-F5
3. change code, save file
4. CTRL-F5
5. etc...

However, right now the steps are:

1. change code, save file
2. Stop debugging instance <<<<<< 
3. CTRL-F5
4. change code, save file
5. Stop debugging instance <<<<<<
6. CTRL-F5
7. etc...

haneefdm, 2022-04-21:
I doubt that. Even in Eclipse, it kinda works the same. It *has* to pretty much everything we are doing.

If you want to just flash the device, then create a task for it. Or create a launch configurations that will exit gdb (`quit`) in the `postLaunchCommands` or `postStartSessionCommands`

haneefdm, 2022-04-21:
We were never designed to just program or erase the device. Those are not debug operations. But you can create launch configurations that can achieve what you want.

haneefdm, 2022-04-21:
I am tempted to lock this issue as it is no longer `issues` we are talking about.

EmanuelW, 2022-04-22:
If it is of any use, here is a complete example of the launch task I talked about. It should behave as you wish @francisjeanneau 

```
{
	"cwd": "${workspaceFolder}/programFolder/",
	"executable": "${workspaceFolder}/programFolder/Debug/program.elf",
	"name": "Program target board",
	"gdbPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gdb.exe",
	"request": "launch",
	"type": "cortex-debug",
	"runToEntryPoint": "",
	"showDevDebugOutput": "parsed",
	"servertype": "external",
	"gdbTarget": "localhost:3333",
	"preLaunchTask": "CopyBinaryToServer",
	"postLaunchCommands": [
		"monitor reset run",
		"detach"
	]
},
```

And since I use remote debugging, I trigger a preLaunchTask that copies the binary to the server first (for the benefit of the GDB server), it should not make a difference in this case but I include the snippet from tasks.json here as well.

```
{
	"label": "CopyBinaryToServer",
	"type": "shell",
	"group": "build",
	"command": "scp ${workspaceFolder}/programFolder/Debug/program.elf user@IP:./openocd/",
},
```

@haneefdm For me I do not have any issue closing this, and feel free to delete this comment if it is too out of scope.

haneefdm, 2022-04-22:
@EmanuelW Thanks for clarifying and posting your launch.json. This thread was closed long time ago but we allow comments in case there is a reason to re-open *for the original issue*. What I don't expect is to take an old/closed thread and morph it into something else.