Marus/cortex-debug #446: JLinkGDBServerCL.exe leaves icon tray icons behind with cortex-debug

pearson, 2021-06-21:
On Windows 10, when a cortex-debug session completes, the icon tray icon for JLinkGDBServerCL.exe is left behind. After multiple debug sessions the icon tray can look the screenshot below. When the mouse is moved over these icons they disappear, which is typical of orphaned icons from killed processes.

Perhaps cortex-debug is not closing the GDB session properly in this case, and the GDB server process is being killed instead of being told to exit?

This becomes a problem when using the  JLinkGDBServerCL.exe icon to launch a web browser connection to receive more information from the debugger. The web connection port changes for each launch instance, so it is useful to have the icon visible.

[image]

haneefdm, 2021-06-22:
Which launch type is it? attach or launch? The process of exiting is a bit different between the two.

Also, this must be something new because this did not happen before (AFAIK) and Cortex-Debug hasn't changed in a while.

pearson, 2021-06-22:
Sorry, @haneefdm, it seems that this also happens when debugging through PlatformIO. I didn't think it did, or at least I don't remember it doing so, but it is definitely doing it now.

Anyway, it looks to be a SEGGER problem.

closed 2021-06-22 by pearson

haneefdm, 2021-06-22:
Killing gdb/server was the last resort. We first ask gdb to detach from the server and exit gracefully.

JLink server and others do not have any way to send an exit command like `monitor exit` so we have to kill it after a timeout if not already dead by gdb disconnecting. Only way is to send a kill command (equivalent to kill(SIGTERM) C-API). SIGTERM is however a **request** which can be ignored.

Can you try the option `-singlerun` on JLink gdb server? By default, it hangs around waiting for the next gdb connection but we had asked it to die which seems to be ignored on Windows.

Also, try with and without SWO and with and without `-singlerun`. Maybe the SWO port needs to be closed as well -- which I am sure we are already doing.

There is another experiment we can run if single run does not work. We should probably do this regardless.

pearson, 2021-06-22:
I didn't mention that it was "launch" that I'm using.

A kill command would definitely match the orphaned icons that I'm seeing. Typically they appear from crashed applications or ones that have been killed through the task manager.

I tried `-singlerun` but it did not seem to change anything. There was one try where the icon seemed to disappear correctly, but that did not repeat in another ~10 tries. The `-singlerun` parameter did appear in the debug console, so it appears to have been set correctly.

I did pull out the `swoConfig` block, but to no effect. AFAIK, this has been a problem since I started using cortex-debug, and I only started with SWO over the past couple of days.

What's the next experiment? 😃

pearson, 2021-06-22:
Also, where is that "gdb disconnecting" timeout in the code? I could tweak it on my end and see what happens.

haneefdm, 2021-06-22:
The disconnect/stop/detach/kill code is here...

https://github.com/Marus/cortex-debug/blob/5e441eca69f1d014185bb9aa4f343fca50d3c7e2/src/backend/mi2/mi2.ts#L232

It passes a `-proc.pid`. This may be a problem on Windows. On Unix it means kill process group. Since you are on Windows, can you remove the `-` sign and see what happens?

That is the experiment. In the function `stop`, you will see a timeout and we ask nicely for gdb to wrap up and WE wait for a second and then kill gdb. That does not kill the server though although a well-behaved one should exit normally.

Notice that we use the kill(...) function but that is actually a gentle one as well. It is unfortunate that is named thusly. Its actual functionality is to send a signal (SIGTERM) to the process. Sending SIGKILL or SIGABRT would have been killing forcefully. I don't understand how we can have orphaned icons and the process be terminated though.

If the above experiment does not work, we have one more thing to try

haneefdm, 2021-06-22:
Btw, when the icon hangs around, what does the task manager say about the JLinkGDBSeverCL.exe process? Can you see it? Still running?

And, btw somewhere in the JLink GDB server documentation, it says to use Control-C to exit/kill the server which is what we are trying to do programmatically. SIGTERM is more or less the equivalent of Control-C.

What version of JLink SW are you running. Wondering if this was a recently introduced issue.

pearson, 2021-06-22:
I'm using J-Link v7.22b which is the latest. I was also using the one version previous to v7.22b.

The JLinkGDBSeverCL.exe process, and an associated JLinkGUIServer.exe process, vanish after the debugging "stop" button is pressed.

I've been digging through the code, and I don't think the the server is being killed as you might think.  I've made several changes but the debugger process is always killed right away:

- I commented out all of `stop()` in `mi2.ts`. I started by commenting out the timeout, but eventually did the whole thing.
- I commended out all of `exit()` in `server.ts`.
- I set a much higher timeout in `disconnectRequest`, at line 809 of `gdb.ts`. It seems that the timeout set here, with its 50ms timeout, would exit the server before a proper disconnect.
- I commented out the contents of `tryKill()` from `mi2.ts`.

It appears that the debugger is being killed from somewhere else, but I'm not sure where that might be.

To make it a bit more difficult, `console.log()` isn't outputting for me. (I'm trying to track that problem down, but no luck so far.) The `this.log()` works within `mi2.ts`, but of course it doesn't work outside of a EventEmitter subclass.

pearson, 2021-06-22:
Just in case this is relevant, the following appears after pressing the debug "stop" button:
```
[2021-06-22 14:09:38.481] [renderer27] [trace] CommandService#executeCommand workbench.action.debug.stop
[2021-06-22 14:09:38.539] [renderer27] [error] read ECONNRESET: Error: read ECONNRESET
	at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
	at TCP.callbackTrampoline (internal/async_hooks.js:131:14)
```

haneefdm, 2021-06-23:
You can see the console.log() output in the parent VSCode's Debug Console where you launched the extension host. None of those timeouts were ever needed with well-behaved servers.

When you have such zombie icons, you said you are getting new tcp ports. This means the old ports are still in use and that means the gdb-server is still running. Can you check with the Task Manager?

pearson, 2021-06-23:
I *should* be able to see the console.log() output in the parent VSCode's Debug Console, but I don't However, if I create a new VSCode extension project and test it from there, I do get that output. I'm not certain what is going wrong there.

What I meant by the ports is that JLinkGDBServerCL.exe will start up a web server for the "J-Link Web control panel". Clicking the icon for JLinkGDBServerCL.exe opens a web browser to the correct port number. I have seen it be at different port numbers, although I can't seem to replicate that at the moment.

The JLinkGDBServerCL.exe task definitely exits when I press the debug "stop" button, though. I'm double-checking that right now, with Task Manager open to the Details page, sorted by name, and "JLinkGDBServerCL.exe" vanishes when the debug "stop" button is pressed.

haneefdm, 2021-06-24:
I hate to ask you this Chris. Reboot? And see if the console.log() works as expected. I have seen this once but could not figure out what did it.

But, I don't understand if `JLinkGDBServerCL.exe` EXITS and you still see its icon. Or it is Windows and its generous latest update(s).

pearson, 2021-06-24:
> I hate to ask you this Chris. Reboot? 

LOL! You must work for my cable internet provider. 😄  

I'll have another look at it after a reboot. 

> But, I don't understand if `JLinkGDBServerCL.exe` EXITS and you still see its icon. Or it is Windows and its generous latest update(s).

I'm still thinking that the process is being killed in some way, since leaving an icon tray icon behind is exactly what happens when a process is killed (or crashes.)

What I find odd is that the `JLinkGDBServerCL.exe` process is still terminating after all that cortex-debug code is commented out. It's as if VSCode is shutting it down immediately when that stop button is pressed, and it isn't relying on cortex-debug to do it.

haneefdm, 2021-06-24:
Ummm. VSCode has no idea about our gdb-server so it cannot possibly do it.

Normally, a gdb-server terminates after a gdb disconnect which is what we are doing. All the timeouts/kills are for servers who do not follow the spec ... or it wants to (or instructed to) remain open after the last connections

pearson, 2021-06-24:
Hmm... I thought I'd commented out anything that was involved in shutting the connection down. I'll have another look at it. (After a reboot)