Marus/cortex-debug #448: Peripheral view spuriously expanded

dzarda, 2021-06-28:
When starting a debug session, some sections in the peripheral view get expanded. They were collapsed before:

[image]

This is a significant time waste.

I had expanded these sections before and it seems like the previous opened state is somehow remembered and applied on every subsequent start. Only specific sections in the SVD are expanded.

dzarda, 2021-06-28:
Ideally the peripheral view should stay the same between debug sessions.

dzarda, 2021-06-28:
(And I hope it's not PlatformIO screwing with me again 😊)

haneefdm, 2021-06-28:
@dzarda I doubt PlatformIO has anything to do with it :-)

Cortex-Debug remembers the state for peripherals window in the file `.vscode/.cortex-debug.peripherals.state.json`. Perhaps we are not updating the file correctly. This file is read after the SVD file is loaded successfully and saved at the end of a debug session.

Can you check the file contents to see what is going on?

There is one caveat that I can think of with multi-root workspaces that may be preventing it from working properly if you have multiple `.vscode` directories and Cortex-Debug is reading from one place but writing to another.

dzarda, 2021-06-28:
No such file is present

dzarda, 2021-06-28:
Oh it appears I'm not using this extension, but had it installed. It's platformio fault again. closing

closed 2021-06-28 by dzarda