Marus/cortex-debug #450: JLink configuration performs a unwanted reflash

eve-cxrp, 2021-07-01:
Hi,

I'm trying to launch a debug session with a JLink adapter. But every time I do this it also is comparing the flash content of the MCU and reflashes the application which I do not want. How do I prevent this from happening? 

Here is my current configuration:

```
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug: MCU",
            "cwd": "${workspaceRoot}",
            "executable": "file.elf",
            "serverpath": "/usr/local/bin/JLinkGDBServerCL",
            "servertype": "jlink",
            "device": "NRF52840_XXAA",
            "interface": "swd",
            "serialNumber": "", //if you have more than one J-Link probe add the serial number here 
            "serverArgs": ["-speed", "1000", "-novd"],
            "runToMain": true,
            "armToolchainPath": "${workspaceRoot}/tools/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin",
            "svdFile": "${workspaceFolder}/file.svd",
        },
```

hwmaier, 2021-07-06:
This is not really an issue with this plugin I believe.

This behaviour is controlled by the J-Link DLL (part of Segger's J-Link toolchain). If the ELF file has changed, you would have to do a re-flash as the binaries won't match and the debugger would be out of sync with the device. The J-Link DLL compares the ELF file with the current flash content, it won't re-flash if there are no changes.

The J-Link tools have a lot of settings you can configure in so called settings and script files, so best to consult the relevant documentation on Segger's web site.  I remember  seeing some settings relating to the comparing behaviour, but you would need to research that for yourself. You can pass a scriptfile using the `jlinkscript` option and a settingfile using `serverArgs` `-settingsfile` option in your launch configuration.

Hope that helps a bit to solve you issue.

haneefdm, 2021-07-06:
You may also want to contact Nordic. Silicon vendors sometimes provide script files (that come with JLink SW) that have the proper memory regions defined and some logic to program/verify the device. Make sure that you are using the correct device name.

eve-cxrp, 2021-07-06:
Well if do it by hand the MCU is not flashed. Only if I run it through the cortex debug. So something is off here.

My manual steps are.

 Consol 1:
`JLinkGDBServerCL -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52840_XXAA -speed 1000 -novd`

Console 2:
```
arm-none-eabi-gdb

> file file.elf
> target remote :50000
> monitor halt
> monitor reset
> c
```

Works like expected.

eve-cxrp, 2021-07-06:
I have run the Cortex Debug session with logging enabled. Here is the log until it suddenly decides to reupload the image.

```
SEGGER J-Link GDB Server V7.50 LogFile
Logging started @ 2021-07-06 13:19
03-00000000-00-00000000-00AA: Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52840_XXAA -speed 1000 -novd -logtofile -log file.log

03-00000000-00-00000000-01AC: -----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              on
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms

03-00000000-00-00000000-0150: ------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 NRF52840_XXAA
Target interface:              SWD
Target interface speed:        1000kHz
Target endian:                 little

03-00000000-00-00000000-0001: 

03-00000000-00-00000001-0018: Connecting to J-Link...

02-00000000-00-00000039-0049: TCD0D000 000:043.006   Firmware: J-Link V10 compiled Jun 29 2021 16:11:46
02-00000000-00-00000040-0049: TCD0D000 000:043.557   Firmware: J-Link V10 compiled Jun 29 2021 16:11:46
02-00000000-00-00000040-003D: TCD0D000 000:043.846   Decompressing FW timestamp took 190 us
02-00000000-00-00000043-0027: TCD0D000 000:047.257   Hardware: V10.10
02-00000000-00-00000043-0025: TCD0D000 000:047.325   S/N: 600111139
02-00000000-00-00000043-0022: TCD0D000 000:047.342   OEM: SEGGER
02-00000000-00-00000043-0045: TCD0D000 000:047.357   Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
02-00000000-00-00000046-0042: TCD0D000 000:050.136   TELNET listener socket opened on port 19021
02-00000000-00-00000046-0030: TCD0D000 000:050.258   WEBSRV Starting webserver
02-00000000-00-00000046-0043: TCD0D000 000:050.310   WEBSRV Webserver running on local port 19080
02-00000000-00-00000046-002E: TCD0D000 000:050.329 - 45.529ms returns "O.K."
03-00000000-00-00000046-0015: J-Link is connected.

02-00000000-00-00000047-0047: TCD0D000 000:050.509 JLINK_ExecCommand("device = NRF52840_XXAA", ...). 
02-00000000-00-00000053-005A: TCD0D000 000:056.801   XML file found at: /Applications/SEGGER/JLink_V750/JLinkDevices.xml
02-00000000-00-00000054-005F: TCD0D000 000:058.143   /Applications/SEGGER/JLink_V750/JLinkDevices.xml evaluated successfully.
02-00000000-00-00000076-0037: TCD0D000 000:080.349   Device "NRF52840_XXAA" selected.
02-00000000-00-00000078-002C: TCD0D000 000:081.671 - 31.148ms returns 0x00
02-00000000-00-00000079-0031: TCD0D000 000:082.943 JLINK_GetFirmwareString(...)
02-00000000-00-00000079-001E: TCD0D000 000:082.978 - 0.036ms
03-00000000-00-00000079-0033: Firmware: J-Link V10 compiled Jun 29 2021 16:11:46

02-00000000-00-00000079-002F: TCD0D000 000:083.044 JLINK_GetHardwareVersion()
02-00000000-00-00000079-002D: TCD0D000 000:083.062 - 0.018ms returns 101000
03-00000000-00-00000079-0011: Hardware: V10.10

02-00000000-00-00000079-0022: TCD0D000 000:083.110 JLINK_GetSN()
02-00000000-00-00000079-0030: TCD0D000 000:083.126 - 0.017ms returns 600111139
02-00000000-00-00000079-002C: TCD0D000 000:083.143 JLINK_GetOEMString(...)
03-00000000-00-00000079-000F: S/N: 600111139

03-00000000-00-00000079-002F: Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB

02-00000000-00-00000079-002B: TCD0D000 000:083.208 JLINK_GetHWStatus(...)
02-00000000-00-00000080-0028: TCD0D000 000:083.467 - 0.262ms returns 0
02-00000000-00-00000080-002C: TCD0D000 000:083.502 JLINK_EnableSoftBPs(ON)
[... 599 lines cut ...]
02-00000000-00-00001306-0038: TD331000 001:309.991   CPU_ReadMem(4 bytes @ 0xE0001000)
02-00000000-00-00001307-001F: TD331000 001:310.540 - 76.113ms
01-0000000A-00-00001307-0028: $526573657474696e67207461726765740D0A#b9
03-00000000-00-00001307-0011: Resetting target

00-0000000A-00-00001340-000C: $Xc200,0:#e3
01-0000000A-00-00001340-0006: $OK#9a
00-0000000A-01-00001341-010E: 2458633230302c3130303a401c0220c1bd0200b759070025be020025be020025be020025be020000000000000000000000000000000000fdba020025be020000000000a5ba0200f1b3020081bd02006792070081bd020081bd020081bd020081bd020081bd020081bd02002f92070081bd020081bd02004d92070081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd020081bd0200233830
03-00000000-00-00001341-002B: Downloading 256 bytes @ address 0x0000C200

02-00000000-00-00001341-0041: TD331000 001:344.461 JLINK_WriteMem(0x0000C200, 0x100 Bytes, ...)
02-00000000-00-00001341-0051: TD331000 001:344.476   Data:  40 1C 02 20 C1 BD 02 00 B7 59 07 00 25 BE 02 00 ...
02-00000000-00-00001341-002A: TD331000 001:345.021   completely In flash
02-00000000-00-00001341-002C: TD331000 001:345.035 - 0.575ms returns 0x100
01-0000000A-00-00001341-0006: $OK#9a
00-0000000A-01-00001341-0800: 2458633330302c336634303a01f0ff01107d0a2bdb10f0070f08d010f8013b013a8b422dd010f0070f42b3f6d1f0b441ea012141ea014122f007047ff00007007d03f0e80256083c85ea010586ea010685fa47f5a3fa87f586fa47f6a5fa87f68eb9eed1f0bc01f0ff0102f0070232b110f8013b013a83ea010313b1f8d10020704701387047002d06bf35 [... cut ...] 7287ed2dfe800f0040404046a6a6a00564ce668b6f1807f04bf6478002c02d0022d05d03ce04ff412716c2000f073fd4fea8c0404f18046acf104047d0406a40d04f1804906f5004609f53049007d04012d02d0022d11d020e0bcf1040f05d2d6f840a1baf1000f18d104e0d9f840a1baf1000f12d14ff000440fe0bcf1040f05d2d6f840a1baf1000f07d104e0d9f840a1baf1000f01d144f0804414f0804fd5d1334c00f1100c07fa0cfcc4f808c38000204452b10764d0f840658d1b25f07f45aa4298bfb5f5800f43d2c0f84015233961
03-00000000-00-00001342-002D: Downloading 16192 bytes @ address 0x0000C300
```

haneefdm, 2021-07-06:
There was no `load` in the command set you sent. Cortex-Debug does a `reset, halt` followed by a `download`. This is expected to just work. Exact commands below

```
            '-interpreter-exec console "monitor halt"',
            '-interpreter-exec console "monitor reset"',
            '-target-download',
            '-interpreter-exec console "monitor reset"'
```

Note that we *have* to use [gdb machine interface](https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_228.html) so we can parse replies back.

You can replace these commands using `override...` in `launch.json` but what we have, works for 99% of the devices and I am wondering why it doesn't for you. There is really no other magic. Sure you can replace these commands and omit the download but I would investigate with JLink and your device vendor why this is not working. As @hwmaier mentioned, If there is ever a mismatch between the elf file and the programmed device, you will never know why things are not working and it can fail in subtle ways.

Perhaps an `"request": "attach",` is more to your use case. In this case, you program the device and hit the reset button and then request an `attach` to a running program.

closed 2021-07-06 by haneefdm

eve-cxrp, 2021-07-06:
It is working for me. But I do have the following problem.

Step 1. flash merged.hex -> contains bootloader + application
Step 2. launch debugger with elf file which only contains the applications.

Now JLink suddenly realizes that the elf file which I have provided has no bootloader and reflashes the MCU with the new elf file. Although the binary has not changed. At least this is my current interpretation of this situation. 

The problem, this reflash causes that the bootloader suddenly no longer sees a bootable image. So I am not able to jump to the application to debug it.

eve-cxrp, 2021-07-06:
> You can replace these commands using override... in launch.json

Thanks for the hint!!! This did the trick.

```        
{
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug: App",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceFolder}/file.elf",
    "serverpath": "/usr/local/bin/JLinkGDBServerCL",
    "servertype": "jlink",
    "device": "NRF52840_XXAA",
    "interface": "swd",
    "serialNumber": "", //if you have more than one J-Link probe add the serial number here 
    "serverArgs": ["-speed", "auto", "-novd", "-nogui"],
    "runToMain": true,
    "armToolchainPath": "${workspaceRoot}/tools/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin",
    "overrideLaunchCommands" : [
        "monitor halt",
        "monitor reset",
    ]
}
```

haneefdm, 2021-07-06:
I have a similar situation but I use a slightly different method.

```
            "preLaunchCommands": [
                // Program via the hex file to get complete coverage
                "exec-file ./build/hello-world.hex"
            ],
```
https://sourceware.org/gdb/onlinedocs/gdb/Files.html

You can also use other formats if supported by JLink. My build produces an elf and a hex file at the same time. The elf file does not have all the info required.

Above changes the actual file being downloaded to the device by gdb (but keeps symbols from the elf file) and keeps everything in sync. I do not use the `overrideLaunchCommands`. Whatever works for you so long as you understand the consequences.

When you first posted the issue, I did not realize what you were trying to do and why it would not work.