Marus/cortex-debug #451: Error on cross ARM toolchain on WSL2

DanieleNardi, 2021-07-05:
Hello,
I'm trying to debug an application on my board using VSCode from Win10 PRO, using a compiler installed in Ubuntu LST 20.04 on WSL 2, but i get the following error message: `vscode failed to launch GDB localhost:50000 connection timed out (from target-select extended-remote localhost:50000).`
Launch configuration is the following:
```
"configurations": [
{
    "name": "Cortex Debug on WSL",
    "cwd": "${workspaceRoot}",
    "executable": "./build/Application_Debug/Application_Debug.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "jlink",
    "serverpath": "/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
    "device": "MK20DX256xxx10",
    "interface": "jtag",
    "runToMain": true,
    "armToolchainPath": "/home/dnardi/gcc-arm-none-eabi-6-2017-q4-major/bin"
}]
```
Debug console output is the following:
```
Please check OUTPUT tab (Adapter Output) for output from /mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe
Launching server: "/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe" "-if" "jtag" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MK20DX256xxx10"
Launching GDB: "/home/dnardi/gcc-arm-none-eabi-6-2017-q4-major/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /home/dnardi/myProject/firmware/build/Application_Debug/Application_Debug.elf...
done.
```

Output is the following:
```
SEGGER J-Link GDB Server V6.94 Command Line Version

JLinkARM.dll V6.94 (DLL compiled Jan  8 2021 13:01:30)

Command line: -if jtag -port 50000 -swoport 50001 -telnetport 50002 -device MK20DX256xxx10
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 MK20DX256xxx10
Target interface:              JTAG
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Feb  4 2021 12:58:41
Hardware: V10.10
S/N: 50118073
Feature(s): GDB
Checking target voltage...
Target voltage: 3.26 V
Listening on TCP/IP port 50000
Connecting to target...

J-Link found 1 JTAG device, Total IRLen = 4
JTAG ID: 0x4BA00477 (Cortex-M4)
Connected to target
```
I ran `netstat -a -b` on Windows prompt to check port 50000 availability: it is available.
I ran `sudo lsof -i -P -n | grep LISTEN` on Ubuntu WSL2 to check port 50000 availability: it is available.

If I run JlinkGDBserverCL manually, then server is available on Windows side, but not on Ubuntu/WSL2 side, so I tried to:
1. enable connection adding a rule in Windows defender firewall for ports 50000-50010
2. adding `"gdbTarget": "172.xxx.xxx.xxx:50000"` (IP got from resolv.conf nameserver) to launch.json
3. adding `"serverArgs": ["-nolocalhostonly"]` to launch.json.

I still get the same error message (still not visible on lsof list). It looks like the gdbTarget option is ignored.

I'm out of options right now, do you have any other input or suggestion?

Thank you very much!

DanieleNardi, 2021-07-05:
If I run jlinkgdbservercl manually with `"/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe" "-if" "jtag" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MK20DX256xxx10" "-nolocalhostonly"` and launch manually gdb with `"/home/dnardi/gcc-arm-none-eabi-6-2017-q4-major/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2" "build/Application_Debug/Application_Debug.elf"`, then on GBS prompt i launch command `terminal remote 172.xxx.xxx.xxx:50000`, then GDB connects to the GDB server.
So, I guess there is something in the middle is not workink within the extension.
GDB version is GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q4-major) 7.12.1.20170417-git.

mpekurin, 2022-01-08:
WSL and Windows don't share the localhost (127.0.0.1) so you can't connect to the gdb server that's running in Windows from WSL via `localhost:xxxxx`. As you've written, `172.xxx.xxx.xxx` (which doesn't work for me) or the local ip of your machine (usually `192.168.0.xxx`) should be used. While there is no complete support for WSL yet, as a temporary solution you can launch the server via a task and use
```
"servertype": "external",
"gdbTarget": "xxx.xxx.xxx.xxx:50000",
```
in your configurations in `launch.json`.

haneefdm, 2023-04-14:
With USBIPD support many people are using USB from WSL2 natively. So, I am going to close this.

closed 2023-04-14 by haneefdm