Marus/cortex-debug #455: Can not debug this extension.

sparkertim, 2021-07-23:
How to debug this extension? I had added nios2-gdb-server support for myself, but I can't debug the ts file, such as mi2.ts or jlink.ts or ublast.ts (added by myself, register it at gdb.ts, works fine.)
The breakpoint not work when I use "Launch extension" or "Extension + Debug Server". so, how can i make it stop at the breakpoint in these files? Can anybody give me some advice? thanks.

haneefdm, 2021-07-23:
You are on the right path. Use "Extension + Debug Server" (which are two separate processes) and in the firmware `launch.json` add the following. Now, the extension/VSCode will use your version of the backend instead of the production/installed version. This is the standard way of debugging debugger backends in VSCode

```JSON
            "debugServer": 4711
```

haneefdm, 2021-07-23:
**Warning**: There are a whole bunch of changes coming soon and will conflict somewhat with what you are doing. Most of my changes are on 

https://github.com/Marus/cortex-debug/tree/origin/rtt

A large number of files have been modified and refactored.

sparkertim, 2021-07-23:
> **Warning**: There are a whole bunch of changes coming soon and will conflict somewhat with what you are doing. Most of my changes are on
> 
> https://github.com/Marus/cortex-debug/tree/origin/rtt
> 
> A large number of files have been modified and refactored.

I got it! thanks!

closed 2021-07-23 by sparkertim