Marus/cortex-debug #458: Can't read some peripheral registers on NXP K81

vome101, 2021-07-29:
I'm failing to see some peripheral registers while debugging. In the picture below, I'm trying to read the SCGC2 register ( 0x4004802c) but the JLink log shows it didn't happen. There is a lot of failed memory read and it eventually stops at 0x40048011. 

[image]

I have a log from another IDE and it always read one byte at a time instead of big chunks (104 bytes).

[image]

haneefdm, 2021-07-29:
Can you compress and attach the SVD file? Unless the registers are weird sizes (not a multiple of 4) I don't know why it would read those odd number of bytes.

vome101, 2021-07-29:
[MK81F25615.zip](https://github.com/Marus/cortex-debug/files/6904014/MK81F25615.zip)

I attached the SVD file. It is trying to read from 0x40048004 to 0x40048068. This is 0x64 or 104 bytes. 

[image]

haneefdm, 2021-09-09:
I don't see anything wrong with the svd file, only way to debug is to see which particular address it is failing at. You can use the MemoryViewer (or gdb-command-line in Debug Console) and perhaps do a binary search as to where it is failing and then we can look at why?

You can invoke the `Cortex-Debug: Memory Viewer` from the `Command Pallete`. Using gdb commands may be faster.

vome101, 2021-09-09:
I think the problem occurs when the extension tries to read a memory address that is not valid. In this example, 0x40048014 isn't a valid register. The output log shows it stopped there. I verified with memory viewer that 0x40048014 can't be read. The SVD file doesn't define 0x40048014. 
The extension tried to read 0x40048004 to 0x40048068 in one command. If it reads each register in separate commands and skip non-defined registers, I don't think this issue wouldn't occur.

haneefdm, 2021-09-09:
Thank you for getting back to me.

With all the other vendors and even with many NXP users, this is not a problem. If you look at the beginning (size) of the peripheral, it assigns a very large block and we used to read the entire block. What is normal behavior is for most ARM devices is to return zeros but not cause a read-failure when accessed via the debug port.

We can provide an option not to read areas where no registers are defined but first I want to make sure that this is the only reason. This was not reported by anyone for years. Reading a register at a time, though makes single stepping unbearable if you have large peripherals.

Can you confirm with NXP if this is normal behavior? I mean they can't do anything about it now, but this is rather strange if that is their policy.

I will look into adding the option to not read empty regions

vome101, 2021-09-09:
I checked another NXP chip, RT1052. It returns 0 for undefined registers. It looks like the K81 chip is a special case.

haneefdm, 2021-09-09:
Thanks for checking that for me. We are exposed to soo many different chips that this was highly unusual.

It may only be some of the Kinetis Series (originally Freescale?)  but you found one family at least.

I will add the feature so you have some control over how gaps in register addresses are handled. Where there is no gap, it will still be a combined read. Our current behavior stays though.

Will be making another release in 1-2 weekends.

vome101, 2021-09-09:
Thanks for looking into this. Looking forward to the update.

haneefdm, 2021-09-10:
Pushed changes for the next release. Reopen if needed

closed 2021-09-10 by haneefdm

vome101, 2021-09-22:
This issue is fixed on 0.4.4 after I add "svdAddrGapThreshold": 0, to launch.json file. Thanks.

haneefdm, 2021-09-28:
We don't use JLink readmem of any kind. We only talk to GDB asking for memory reads. GDB in turn asks JLinkGDBServer which does whatever it takes to read the memory. We never talk to JLink directly.

My question is why is JLink failing to read memory. It is a valid command.

https://wiki.segger.com/J-Link_Commander#Mem

So, we need to focus on addresses and data sizes. Do the addresses or sizes look wrong? I don't have this device, only you can tell. Using gdb commands, in the `Debug Console` try to read those same memory locations/sizes as see what happens

https://sourceware.org/gdb/current/onlinedocs/gdb/Memory.html

haneefdm, 2021-09-28:
@vome101 Have you contacted SEGGER or NXP for an explanation? We use gdb+JLinkGDBServer and that combination *has* to work. You can leave out Cortex-Debug and use both tools from the command line and reading of memory from gdb *has* to work. Or, there is a bug in JLinkGDBServer.

vome101, 2021-09-28:
I also used JLink commander. It shows mem failed to read 8 bytes while mem8 succeeded.
[image]

Each register has a size: 8, 16, 32. Do you use this size to do mem8, mem16, mem32 read?
[image]

I see that GDB has an option for size.
[image]

haneefdm, 2021-09-28:
We already do use `read-bytes`. We do not read registers. There is no such interface to gdb. Only read memory. You can see that in the commands we send to the GDB. In your launch.json you can see
```
"showDevDebugOutput": true
```
Btw, we HAVE to use the gdb machine interface commands are the return values are parseable and there is a spec for it, unlike regular gdb commands.

https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI-Data-Manipulation.html#GDB_002fMI-Data-Manipulation

We use the command 
```
-data-read-memory-bytes address count
```
So, what you do with JLink commander does not apply here, as we cannot use that in any way. We HAVE to use the JLinkGDBServer via gdb.

SEGGER has to help you with this. They are the ones who use ReadMem, etc. Not us or GDB.

vome101, 2021-09-28:
I'll ask Segger about why it is using JLINK_ReadMem instead of JLINK_ReadMem8 for bytes read. 

Can you make this improvement? If GDB fails to read a chunk of bytes, it will try to read one byte at a time for that chunk. In the below example, it failed to read 18 bytes at 0x4007C000. Then it read up to 0x4007C003 but stopped instead of continuing up to 0x4007C012.

[image]

haneefdm, 2021-09-28:
Sorry @vome101 , but I have to decline. That is a lot of work and error-checking, testing for us. Besides, we will have no idea why it failed and then we would fail slowly again. And it will go on..

It appears that this device was never made for a tool like ours.