Marus/cortex-debug #459: Failed to launch OpenOCD GDB Server: Error: spawn

jrhug, 2021-08-06:
I have search through the issues and I am still unable to get OpenOCD to work with VS Code and cortex-debug.  My issue is identical to:  https://github.com/Marus/cortex-debug/issues/281#issue-607868097
Unfortunately, updating my system path to include OpenOCD did not correct my problem.  I have tried several versions of OpenOCD and I get the same issue.  As a hail mary, I attempted to run the debugger in safe mode.  This provided interesting error output. 
[image]

The error output running normal windows is as follows:
[image]

Both applications, openocd.exe and arm-none-eabi-gdb.exe can be launched from the command window without issue.  The utilities are accessible in the designated files.

The following versions of Windows, VS Code, and Cortex-Debug are being used.
[image]

Here is my launch.json settings:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug",
            "cwd": "D:/target",
            "executable": "${workspaceFolder}/target/thumbv7em-none-eabi/debug/target.elf",
            "request": "launch",
            "type": "cortex-debug",
            //"servertype": "external",
            //"gdbTarget": "localhost:50000",
            "servertype": "openocd",
            "searchDir": [
                "D:/boards/target/jtag"
            ],
            "configFiles": [
                "config.cfg",
                "chip.cfg",
                "board.cfg"
            ],
            "runToMain": false,
            "preRestartCommands": [
                "monitor reset",
                "file D:/target/thumbv7em-none-eabi/debug/target.elf",
                "load",
                "target extended-remote :50000",
                "monitor reset halt"
            ]
        }
    ]
}
```
Note, I tried setting the severtype to external and I get the same ENOENT error for the armToolchainPath.  
Here is my settings.json
```
{
    "editor.formatOnSave": true,
    "cortex-debug.openocdPath": "C:/Users/<user>/AppData/Roaming/xPacks/xpack-openocd-0.11.0-1/bin/openocd.exe",
    //"cortex-debug.openocdPath": "C:/Temp/xpack-openocd-0.11.0-1/bin/openocd.exe",
    //"cortex-debug.openocdPath": "openocd.exe",
    "cortex-debug.armToolchainPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin",
}
```
What else can I try to get the cortex debugger to work? 

Thank you in advance for any feedback.

haneefdm, 2021-08-06:
Hmm, this is a head-scratcher. You don't happen to have a user name that has special characters? If you have anything but [7-bit printable ascii](https://sites.google.com/site/tomihasa/ascii-7-bit-printable), then it might be a problem

But you say we also have a problem with gdb. The gdb we run is `arm-none-eabi-gdb` and not plain `gdb` There are sometimes two bin directories in the compiler distribution. Make sure you are using the right directory for the `armToolchainPath`

jrhug, 2021-08-06:
> Hmm, this is a head-scratcher. You don't happen to have a user name that has special characters? If you have anything but [7-bit printable ascii](https://sites.google.com/site/tomihasa/ascii-7-bit-printable), then it might be a problem

The user name is standard ASCII characters; no special characters.

> But you say we also have a problem with gdb. The gdb we run is `arm-none-eabi-gdb` and not plain `gdb` There are sometimes two bin directories in the compiler distribution. Make sure you are using the right directory for the `armToolchainPath`

Correct, I am running the armToolChain gdb.  For completeness, I have pasted my path settings.
[image]

haneefdm, 2021-08-06:
can you do a `dir` on those to directories and paste the output? I don't think it is a security issue because the message says `ENOENT` meaning file does not exist. Or else it would have said `EPERM`.

We don't do anything special. All we do is spawn (NodeJS API) and looks like it is failing.

jrhug, 2021-08-06:
> can you do a `dir` on those to directories and paste the output? I don't think it is a security issue because the message says `ENOENT` meaning file does not exist. Or else it would have said `EPERM`.

[image]

[image]

As an experiment, I cloned the cortex-debug repo so I could try to gather more debug information.  I am having a compilation issue since I am unable to install webpack.  I get a dependency error but, I am not familiar with Node.js.  Is the dependency error easy to address?  Does the attach log file mean anything useful?  [eresolve-report.txt](https://github.com/Marus/cortex-debug/files/6948044/eresolve-report.txt)

> We don't do anything special. All we do is spawn (NodeJS API) and looks like it is failing.

Could it be I have an incorrect version of NodeJS?  I was running NodeJS version 14.6.1.  As a test, I uninstalled NodeJS and tried running the debugger again.  As expected, I get the same error "_Failed to Launch..._".

haneefdm, 2021-08-06:
I doubt NodeJS is wrong. I am using 14.6.0 and I am getting the same problem if I change package.json to upgrade to latest webpack. It still compiles though.

After cloning the report, just run 'npm install', do not install/upgrade any modules. There are too many dependencies and if you change one thing, it leads to another. I will clean it up when I have some time.

jrhug, 2021-08-10:
**Update**: Long story short, I built a new system and loaded a fresh copy of Windows 7.  At first, I had the same issue.  After playing around a bit, I finally got the extension working.  Will provide further details in the future.  Closing this issue now.

closed 2021-08-10 by jrhug

ashish-gawali-adi, 2023-03-28:
If you see this, please do share the solution to this problem. 
Warm Regards,
Your frustrated about to throw the keyboard out of the window software engineer.