Marus/cortex-debug #461: "monitor reset halt"

r00tc0d3, 2021-08-09:
Hello,

Please change for the external interface "monitor reset halt" to "monitor reset" and "monitor halt"
When using JLink GDB server this will fail because Segger JLink does not support "monitor reset halt". If you split is up it will be compatible with al GDB servers. There I would also suggest to change it everywhere because splitting it up always works but combining monitor command does not work for every GDB server.

But at least for external because there is no way to know what server is used. It is external isn't it :)

haneefdm, 2021-08-09:
But, the best way to use `external` type server is to use the `override...` options in the launch/attach commands sent to gdb/gdb-servers. There is no way we can have one set of commands that satisfied all gdb-servers, especially for `external`.

And no, AFAIK, doing `monitor reset` followed by `monitor halt` is not the same thing in OpenOCD or other gdb-servers. Just doing a `monitor reset` means do a reset followed by a run. So, if I do what you suggest, the halt happens in some random location. In JLink just doing `monitor reset` is enough. because it resets and halts. Totally different meanings

http://openocd.org/doc/html/General-Commands.html

Also, depending on your device in JLink, you may have to supply a reset strategy like `monitor reset 0/1/2` (see section 7.10 of JLink software)

haneefdm, 2021-08-09:
closing this issue because it will be a breaking change and there is really no good way to have one size fit all for external servers. We have trouble enough with known supported servers.

closed 2021-08-09 by haneefdm

r00tc0d3, 2021-08-10:
Thats Ok. I did not know about the override yet.
I personlay don't understand why openocd or segger or whatever are al that stubborn and make all there own implementation and behavior. But that out our control, so we will make a lauch entry for every server there is ;-) Thanks to overrride I can :) tweak arround.