Marus/cortex-debug #464: Task always launches when extension is activated

trond-snekvik, 2021-08-13:
After the most recent release, the Cortex debug extension keeps launching and revealing a task window that says "Waiting for gdb server to start..." when being activated. 

If this extension is being activated on more events than the configured debug resolution event, this is very disruptive. 
Extensions will always be activated when an extension that depends on them is activated, so if a user has an extension installed that lists this extension in `extensionDependencies`, the window will pop up as soon as the other extension is activated, making integration a bit awkward.

I'd suggest moving this task launch to somewhere other than the activate function, and potentially changing the task presentation  configuration to not reveal itself by default. A configuration setting to disable this behavior altogether would also be handy.

haneefdm, 2021-08-13:
Yes, that is my bad. I left it in there but it was supposed for Debug purposes. The automatic showing of the gdb-server terminal will not be there in the next release. Thanks for reporting this. Also, this only happens once for the first session. Definitely, not every time you launch the debugger. Also, I was trying to mirror the behavior we had before where the 'Adapter Output' window would show itself at the very start.

And no, this window will not move from the Terminal area. It was done to support semi-hosting and there is no other suitable place. And, it needs to look like a Terminal.

About activation, nope, this extension is not activated until you launch a `cortex-debug` session. As disruptive as it is, **we** don't do it and I don't see it on Windows and Mac. The following is our whole activation setup in `package.json` and it has been that way since day one.

```
    "activationEvents": [
        "onDebugResolve:cortex-debug"
    ],
```
So, unless VSCode is doing something .... I don't see how.

closed 2021-08-13 by haneefdm

trond-snekvik, 2021-08-17:
> About activation, nope, this extension is not activated until you launch a cortex-debug session.

haneefdm, 2021-08-17:
> As explained in the initial comment, VS code will forcibly activate this extension if any extension that has it as a dependency is activated

Yes, what VSCode does, makes sense, but what can I do? Now the terminal is created but does not pop up to the top.  The user has to manually expose it. Did you try the latest release? You can execute a "Debug: Focus on Debug Console" command. Generally, uses launch VSCode and stay in there for days.

I cannot remove the important/needed functionality that you are asking me to do. It took me months of planning and work to get this functionality going. I tried to delay the creation of the terminal but it was always too late. VSCode does not have an event for `debugSessionStarting`. Only `debugSessionStarted` by which time, execution has already started. I will take one more shot at it, refactor my code, but I can't test it in your use case.

Another question I have is could you not have Cortex-Debug in `extensionDependencies`. Instead, just let the normal process of launching a debugger activate it. Another way is to have an [`Extension Pack`](https://code.visualstudio.com/api/references/extension-manifest#extension-packs) Do you have a functional dependency on this extension that requires you to add it as an `extensionDependencies`.

Do you mind sharing your repo location?

haneefdm, 2021-08-17:
I have an idea that I want to try but I will not know if that works for you or not. Is there anything you can package from your extension that demonstrates the issue?

trond-snekvik, 2021-08-24:
Sorry for the slow response. I'm not able to share the original extension that triggered this, unfortunately, so I made a minimal extension that demonstrates the behavior: https://github.com/trond-snekvik/vscode-cortex-debug-issue 
A precompiled vsix is available here: https://github.com/trond-snekvik/vscode-cortex-debug-issue/releases/download/v0.0.1/cortex-debug-test-0.0.1.vsix

haneefdm, 2021-09-09:
@trond-snekvik, There is an update that might help your situation. Changes have been pushed to the repo if you want to try it out. Also, there is a `Reset device` button that you could try out -- I believe this was your suggestion.