Marus/cortex-debug #465: Non printing characters and padding in SWO console output lead to premature line wrapping

NuclearLighthouseStudios, 2021-08-13:
When printing characters to the SWO console in console.ts the current position is advanced even when the current letter is non printable.

Since it is common to only send one character inside a 4 byte ITM message and have the remaining three bytes be zeros this leads to messages unnecessarily wrapping after only 20 characters which makes the output very hard to read.

haneefdm, 2021-08-13:
Ummm, more details, please. The SWO output is decoded according to the ARM spec and we wait until a full packet arrives in cosole.ts. That packet parsing code is actually in `core.ts` and has not changed in ages.

See [ARM architecture reference manual Appendix D.2.9](http://mazsola.iit.uni-miskolc.hu/~drdani/docs_arm/DDI0403D_arm_architecture_v7m_reference_manual_errata_markup_1_0.pdf)

But, if you are sending binary data, maybe `console` is not the best way to look at it. You may want to use decoder type `binary`. Or maybe it is your FW. You are writing 4 bytes instead of 1. For my FW, I get 1-byte packets. Not 4

```C
  ITM->PORT[portNo].u8 = c;
```

Packet sizes can be 1, 2 or 4 bytes long with a one-byte header. SWO is already slow so, sending more than needed is also not good.

Please close this issue if it is not a bug on our side

NuclearLighthouseStudios, 2021-08-13:
Ahh, turns out I was doing a 32 bit write to the stimulus register instead of an 8 bit write. I wasn't aware that the packet length is influenced by the width of the write.
Sorry for the confusion.

closed 2021-08-13 by NuclearLighthouseStudios