Marus/cortex-debug #468: STLink servertype debug fails in launch request

hihigupt, 2021-08-20:
I am using STLink servertype to launch the debugger that downloads and attach to the device. But debugger fails to figure out right PC value and hangs the debugger in invalid PC value. Following are configuration, steps and logs for my setup.

Configuration:
{
            "name": "STM L475E: Local Stlink",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "stlink",
            "runToMain": true,
            "device": "STM32L475VG",
            "postStartSessionCommands": ["monitor reset"],
            "executable": "${workspaceFolder}/b-l475e-iot01a/demos/projects/ST/b-l475e-iot01a/iot-middleware-sample.elf",
            "cwd": "${workspaceRoot}",
            //"preLaunchTask": "Build b-l475e-iot01a"
}

Steps:
1. Erase STM32L475-disco board.
2. Launch debugger that downloads your firmware and attach to it.
3. Debugger now is unresponsive.

Note: make sure device is erase and does not contain any valid firmware.

Logs:

Adapter logs:

STMicroelectronics ST-LINK GDB server. Version 5.8.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target 
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J37M27
Device ID: 0x470
PC: 0x1fff51da
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.23 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Waiting for debugger connection...
Waiting for connection on port 50000...
Accepted connection on port 50000...
Debugger connected
Enter STM32_AppReset() function 
NVIC_DFSR_REG = 0x00000008
NVIC_CFGFSR_REG = 0x00000000
 ------ Switching to STM32CubeProgrammer ----- 
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.7.0-RC1                  
      -------------------------------------------------------------------

ST-LINK SN  : 066BFF3134354D5043061952
ST-LINK FW  : V2J37M27
Board       : STM32L4S5I-DK
Voltage     : 3.23V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x470
Revision ID : Rev Y
Device name : STM32L4Rxxx/STM32L4Sxxx
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M4

Memory Programming ...
Opening and parsing file: ST-LINK_GDB_server_a18240.srec
  File          : ST-LINK_GDB_server_a18240.srec
  Size          : 312648 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sector 0
Erasing memory corresponding to segment 1:
Erasing internal memory sectors [0 76]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:08.703
 ------ Switching context ----- 
Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target 
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J37M27
Device ID: 0x470
PC: 0x1fff51da
ST-LINK detects target voltage = 3.23 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Enter STM32_AppReset() function 
NVIC_DFSR_REG = 0x00000008
NVIC_CFGFSR_REG = 0x00000000
TraceCaptureStart and SWV event set to APP_TRUE 
ST-LINK device status: RUN_MODE

Debug console: 
Launching GDB: "c:\ST\STM32CubeIDE_1.3.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\bin\arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
undefinedc:\ST\STM32CubeIDE_1.3.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from C:\random/b-l475e-iot01a/demos/projects/ST/b-l475e-iot01a/iot-middleware-sample.elf...
0x1fff51da in ?? ()
Program stopped, probably due to a reset and/or halt issued by debugger
2
STM32 Successfully completed reset operation
Note: automatically using hardware breakpoints for read-only addresses.

Based on above it looks like debugger attach to the server before download finishes and points to random PC (PC: 0x1fff51da) and causing debugger to stuck. If, yes is there some post/pre hook that can help to avoid this issue.

haneefdm, 2021-08-20:
Yes, the debugger does attach to the server at the very beginning. We (cortex-debug via gdb) then ask the server to load the program and nothing in gdb moves forward until the server finishes the download. and we are simply waiting for all the initialization commands are executed.

You can enable see the gdb transactions by doing `"showDevDebugOutput": true` in your launch.json

Yes, there are plenty of options for pre... post... override.... In VSCode, open launch.json and start typing and you will see all those options.

Can you change `runToMain` to false and see what happens? It should stop in the reset vector and from there, if you continue, what happens? If it hangs again, it is a problem in the boot code or something else is wrong We have to see how far you can go and what causes the issue. Painful, I know.

> Couldn't determine a path for the index cache directory

BTW, if you are interested, you can get rid of this message if you set an environment variable HOME to be sat value as USERPROFILE.

haneefdm, 2021-08-20:
I see that in your launch.json, you are doing...
```
"postStartSessionCommands": ["monitor reset"],
```
Cortex-Debug has already done a "monitor reset halt". The above will more than likely do another reset but may also continue to execute and may end up somewhere else that we don't expect. Please see the documentation for STLink gdb-server to see the effect of your "monitor reset". If the program does not stop in the reset vector at the start, then many things may not work.

Every gdb-server does this reset in slightly different ways.

hihigupt, 2021-08-20:
Even if I remove this postStartSessionCommands, debug behavior remains same .... I was just trying pre/post hook of server or client, but looks like none helped and debugger hangs on PC address that is printed on Adapter console (i.e PC: 0x1fff51da). Also, please note request is launch.

haneefdm, 2021-08-20:
You have both command lines  (gdb and the server) and all the commands going back and forth between Cortex-Debug and gdb.

Can you run both (gdb and the server) on the command line and see if you can duplicate it and see what is causing it. Not having your exact setup, I would not know how to duplicate it using the `override...` commands. If you succeed please do post back so it can help others.

If you have another debugger that is working, can you see what commands are going back and forth and try to duplicate the same. If it is Eclipse, you can see those same transactions -- I don't remember how to enable the verbose stuff. I think it is a hidden console you have to expose or something.

hwmaier, 2021-08-21:
> But debugger fails to figure out right PC value

It is strange to see an entry point address of `0x1fff51da` which would not be in flash memory on an ST device. Could this be an issue with the way the project is build? Maybe the linker script? Also have a look at your map file to see which address main() and  Reset_Handler() are sitting. With the ST linker scripts the entry point is usually set to `ENTRY(Reset_Handler) `

hihigupt, 2021-08-21:
Things work fine if board already has firmware preinstalled... then I can see address to be valid and debugger works fine.... Now in case when flash has no preimage installed, this issue get surfaced. 

This make me think that gdb client is connecting to gdb server when firmware has not yet been downloaded. In this case PC is 0x1fff51da which is in system memory address space.

hwmaier, 2021-08-21:
On an ARM MCU after a reset the initial PC is loaded from the vector table. So the correctness of the initial PC depends on the vector table to be in flash memory.

I suggest to check your map file to see that the vector table has been linked in and is at the correct address and that Reset_Handler points to something sensible in flash. Example how these entries should look like:

```
 .isr_vector    0x08000000      0x1c8 bsp/board_stm32f746g-disco/libbsp_stm32f746g-disco.a(startup_stm32f746xx.s.obj)
                0x08000000                g_pfnVectors

```
```
 .text.Reset_Handler
                0x080022f8       0x50 bsp/board_stm32f746g-disco/libbsp_stm32f746g-disco.a(startup_stm32f746xx.s.obj)
                0x080022f8                Reset_Handler

```

I you for example forget to link the vector table (part of startup_stm32xxxx.s) then your program will compile and flash to the board successfully but the debugger will not be able to run the code nor being able to set a breakpoint.

hihigupt, 2021-08-23:
@hwmaier I am using binary with valid vector table layout. Following is the first 8 bytes in my binary:

random@:/mnt/c/gitroot/iot_8/iot-middleware-freertos-samples$ hexdump b-l475e-iot01a/demos/projects/ST/b-l475e-iot01a/iot-middleware-sample.bin |head
0000000 **8000 2001** **aba5 0801** 24d5 0800 24d7 0800
0000010 24d9 0800 24db 0800 24dd 0800 0000 0000
0000020 0000 0000 0000 0000 0000 0000 51f1 0800
0000030 24df 0800 0000 0000 52c1 0800 24e1 0800
0000040 abf5 0801 abf5 0801 abf5 0801 abf5 0801
0000050 abf5 0801 abf5 0801 abf5 0801 250d 0800
0000060 abf5 0801 abf5 0801 abf5 0801 abf5 0801

SP=20018000
PC=0801aba5

1. Case it works:
Device already has that firmware and you start the debugger that launches your new firmware (launch = download to device and then attach). In this case debugger is able to debug stuff correctly.

2. Case when it does not work:
Device is fully erased, and then you start debugger that launches your firmware. In this case debugger hangs.

I think above simple steps, should allow you to reproduce it locally.

haneefdm, 2021-08-23:
I have worked with devices where the executable (elf) file does not have **ALL** the information required to properly program and function. Unfortunate, but true. It appears you using the `.bin` file. If possible you can use the same to program the ST device via Cortex-Debug. For a case like that, I did the following.

```
            "preLaunchCommands": [
                // Program via the hex file to get complete coverage
                "exec-file build/Debug/hello-world.hex"
            ],
```
I used the hex file but you can try using the .bin file. Note that my .hex file is an Intel hex file format. Not the one produced by hexdump. Try your .bin file directly instead of the .hex file like I did or use `objcopy` to do a format conversion.

closed 2021-09-09 by haneefdm

pilcherd, 2021-12-27:
@hihigupt Did you ever find a solution to this?  I see the issue was closed. But I am having this same issue.

pilcherd, 2021-12-27:
I see this post by ST which might be related where they state that PEMPTY is not being cleared when downloading the application and starting debugging.  https://community.st.com/s/question/0D53W00000raYd2SAE/intermittent-break-at-address-0x1fff51f4-with-no-debug

I can confirm that if I use STM32CubeProgrammer to program my BIN, it runs. 

Note my setup was working perfectly on a board using STM32L4S5, but we had to move to STM32L4S7 and despite updating all makefiles/LD etc, from within VSCode and Cortex Debug, I'm having this issue.  If I set a breakpoint in Reset_Handler with runToMain set to false, it doesn't make it there.  I have verified BOOT0 is pulled low.  Odd.

haneefdm, 2021-12-27:
Try `breakAfterReset`. Do not use `runToMain` (deprecated) or `runToEntryPoint`. This does not use a breakpoint but your device has to handle a `reset` properly and halt at the reset handler normally.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

pilcherd, 2021-12-27:
Thanks. It appears that once an MCU has been flashed using STM32CubeProgrammer (rather than via cortex debug/gdb/openocd), that everything works ok.   I am able to develop now... weird.  I will try that option on a new board or perhaps try a complete MCU erase and see if I can get it back in that state.