Marus/cortex-debug #472: Breakpoint stops at wrong line

bmcdonnell-fb, 2021-08-28:
As shown in the screenshot, the debugger is stopping at the wrong place. It isn't stopping at the designated breakpoints. The breakpoints are inside an `if` block, so they might not hit, but it is stopping after that block, for unknown reasons.

This is an STM32F4 / Cortex-M4 application, running FreeRTOS. I'm debugging w/ a Segger J-Link, with their software V7.52c.

[image]

haneefdm, 2021-08-28:
Can you check to see the breakpoints information? We ask gdb to set a breakpoint at the line you asked for. But gdb may set it at a line near it. This is true of almost every debugger. Can you compare what is in the breakpoints window vs what gdb reports? You can find out by entering the command `info breakpoints`  in the Debug Console.

You can also ask gdb for the stack/back trace using the `bt` command. We only report what gdb reports.

Make sure your build is up to date and optimizations disabled. gcc still does some optimization when when asked not to. Use `-O0 -Og` just to be sure.

haneefdm, 2021-08-28:
There is one thing our debugger could improve on is perhaps visually  (and temporarily) move the breakpoint to where it actually got set. Not sure if gdb (and VSCode) allows us to do this but I will look. For instance, you can set a breakpoint on a blank line and it will actually get set on the next non-optimized out statement.

bmcdonnell-fb, 2021-08-28:
> We ask gdb to set a breakpoint at the line you asked for. But gdb may set it at a line near it. This is true of almost every debugger. Can you compare what is in the breakpoints window vs what gdb reports? You can find out by entering the command `info breakpoints` in the Debug Console.

Your hypothesis was correct.

> We only report what gdb reports.

GDB is lying?

> Make sure your build is up to date and optimizations disabled. gcc still does some optimization when when asked not to.

It is an up to date debug build.

> Use `-O0 -Og` just to be sure.

I'll try that later, thanks.

haneefdm, 2021-08-28:
> GDB is lying?

You may call it that, but I wouldn't call it lying. It was best effort based on what gcc/ld did. We don't have a reason to lie either.

closed 2021-08-28 by haneefdm

bmcdonnell-fb, 2021-08-28:
Sorry about my phrasing. I jumped to a conclusion following your simplified explanation.

At some level, what is requested is not being done, and that feedback wasn't given to the user (me). I thought you were saying that when cortex-debug asked GDB to set the breakpoint at line 959, that the feedback from GDB to cortex-debug indicated that it had set the breakpoint at that line, whereas in reality it had set it elsewhere.

On further reflection, I imagine maybe the feedback from GDB is that it set a breakpoint "at or near" the requested line--best effort, as you say.

> There is one thing our debugger could improve on is perhaps visually (and temporarily) move the breakpoint to where it actually got set. Not sure if gdb (and VSCode) allows us to do this but I will look. For instance, you can set a breakpoint on a blank line and it will actually get set on the next non-optimized out statement. A bit of work tough

I'd appreciate if cortex-debug is able to improve the feedback in this regard. Would it be appropriate to reopen and use this issue for that? Or start a new one? Thanks.

haneefdm, 2021-09-07:
In the next release, the breakpoint (red dot) will move to the actual location and move back to its original position when the session ends.

bmcdonnell-fb, 2021-09-10:
> the breakpoint (red dot) will move to the actual location

Can you make this happen when the user sets the breakpoint?

> and move back to its original position when the session ends.

I don't think it should.

haneefdm, 2021-09-10:
> Can you make this happen when the user sets the breakpoint?

Nope. No facility to do that and not a good idea either. A debugger cannot insert, delete or change breakpoints in the GUI. It is pretty much a slave to the GUI.

> I don't think it should.

See above. The GUI does this. And, this is correct, IMO

Having said that, this is what every other debugger I know of also does. I would actually get mad if a debugger did that to me.

haneefdm, 2021-09-10:
Btw, VSCode belongs to MS and they control 90+% of how the UI works. We try and conform to an ever-evolving specification/API published by my MS which has zero capabilities for UI-related stuff. Don't think that will change.

https://microsoft.github.io/debug-adapter-protocol/

This spec is followed by dozens of debuggers including MS's own ones.

bmcdonnell-fb, 2021-09-10:
> > Can you make this happen when the user sets the breakpoint?
> 
> Nope. No facility to do that and not a good idea either. A debugger cannot insert, delete or change breakpoints in the GUI. It is pretty much a slave to the GUI.

After the GUI tells the debugger to set a breakpoint, could it query the debugger to find out the actual location of the breakpoint(s)? If so, why would that be a bad idea? I imagine it could allow the developer to see sooner that the breakpoint is not at the intended place, which would be to the advantage of detecting an error at compile time error instead of a runtime.

It's particularly troublesome in cases like my example above, where the breakpoint never hits, because its actual location is in a different code path.

haneefdm, 2021-09-10:
There a many reasons why it is a bad idea. Breakpoints have a life without a debugger even existing.

If I ask for line 'X' stay at line 'X' is how I want it. Linx 'X' may become valid again. Even a file can appear and disappear as part of a build and I don't want my breakpoints messed with.

Please feel free to take your use case to Microsoft to change their GUI,