Marus/cortex-debug #474: Cannot start debug session - debugger thinks OpenOCD shut down, yet OpenOCD is running correctly

FreddieChopin, 2021-08-30:
The setup I have worked perfectly fine with previous version of vscode, however after upgrading my system (Arch Linux) recently, I can no longer start the debugging session with this plugin. It appears that the plugin starts OpenOCD just fine, but then somehow debugger thinks that OpenOCD shut down and it just halts. Yet OpenOCD is working in the background completely fine. When I try to start the debugger again, it obviously fails, because it tries to start OpenOCD again, yet one instance is still running in the background. If I kill it manually and try again, the situation repeats.

This is what I see in the gdb-server terminal in vscode:

```
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server program connected
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
none separate

Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 50000
Info : Listening on port 50000 for gdb connections
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server program closed
GDB server exited. Waiting for next server to start...
```

This is my config file:

```
{
	// Use IntelliSense to learn about possible attributes.
	// Hover to view descriptions of existing attributes.
	// For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
	"version": "0.2.0",
	"configurations": [
		{
			"cwd": "${workspaceRoot}",
			"executable": "./output/STM32F4-module.elf",
			"name": "STM32F407VE-module (with load)",
			"request": "launch",
			"type": "cortex-debug",
			"servertype": "openocd",
			"configFiles": [
				"STM32F407VE-module.cfg"
			],
			"svdFile": "/home/freddie/Elektronika/ARM/cmsis-svd-master/data/STMicro/STM32F40x.svd",
			"preLaunchTask": "Build",
			"postLaunchCommands": [
				"monitor reset halt",
				"thbreak main"
			]
		}
	]
}
```

This are the contents of the OpenOCD cfg file:

```
source [find interface/ftdi/jtag-lock-pick_tiny_2.cfg]
transport select swd
source [find target/stm32f4x.cfg]
reset_config none
```

This exact same setup worked perfectly fine before the update. Please let me know if any additional info is required.

Version: 1.58.0
Commit: 2d23c42a936db1c7b3b06f918cde29561cc47cd6
Date: 2021-07-12T23:51:36.303Z
Electron: 12.0.17
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux x64 5.13.13-arch1-1

plugin version: 0.4.2
OpenOCD 0.11.0

haneefdm, 2021-08-30:
> SERVER CONSOLE DEBUG: onBackendConnect: gdb-server program closed

It appears OpenOCD is crashing/exeited. I don't think this has anything to do with VSCode update. Something else is going on. Can you start the gdb-server (OpenOCD) from the command-line? You can see the exact command-line used in the `Debug Console` window?

FreddieChopin, 2021-08-30:
In the end I have no idea what the problem was, as after a series of downgrades (which did not change a thing) followed by upgrades to latest version everything just started to work normally... It's possible that the application that was crashing was GDB itself, OpenOCD was not crashing - this is one thing I'm sure about in this whole situation.

Sorry for wasting your time, it's good that it is working fine now!

closed 2021-08-30 by FreddieChopin

J-Rios, 2021-09-01:
Hi, for anyone that found this issue in the future...

I was facing a similar issue, seems that GDB closes unexpectly:

```bash
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server program closed
GDB server exited. Waiting for next server to start...
```

First, check if there is any problem when you try to debug manually from command line:

```bash
arm-none-eabi-gdb yourfirmware.elf

arm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

Then, check what is your specific problem. In my case, the problem was a missing package, and just just need to install libncurse. For Debian based:

```bash
sudo apt update
sudo apt install libncurses5 libncurses5-dev libncursesw5-dev
```

Now vscode launch correctly the debug and play/stop floating menu appears.

Regards.

haneefdm, 2021-09-01:
The libncuses problem keeps coming up soooo many times. I am not sure why the ARM gdb installation (or wherever people download gcc/gdb from) is incomplete on Linux and why this is not reported to ARM.