Marus/cortex-debug #475: Debugging with openocd no longer works

arif-kun, 2021-09-06:
Hey debugging using OpenOCD no longer works for some reason.  This is the message I am getting :

SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. Waiting for next server session to start...

I am using **VSCode** on **Windows 11** Beta. Openocd installed through MSYS2. Even tried the xPack openocd binaries.
Tried on my **Manjaro** VM too, same issue.

Using various STM32 Nucleo boards with st-link.

haneefdm, 2021-09-06:
Could you please try running openocd from the command line? You can see the exact command-line used in the 'Debug Console'.

"no longer works". when did it start failing? A release occurred less than a day ago but no changes related to gdb-servers though.

Also, if you are using ST devices, don't you have to use OpenOCD shipped with the Cube tools? Although, it should not matter for this purpose.

arif-kun, 2021-09-06:
> Could you please try running openocd from the command line? You can see the exact command-line used in the 'Debug Console'.
Ok I'll try this.

> "no longer works". when did it start failing? A release occurred less than a day ago but no changes related to gdb-servers though.
It was working till last month. Now I sit again for embedded learning, it no longer works.

I tried debugging on my Manjaro VM using the exact same settings as my host Windows, and there it works fine.  So the extension works. It's either a Windows problem or openocd binaries on Windows problem. So I am closing the issue.

closed 2021-09-06 by arif-kun