Marus/cortex-debug #476: SWO output missing from the output tab

andyinno, 2021-09-06:
Hi,

It is some time that my config (that was working before) did not happen to be working anymore.

Basically in output I could select the SWO output that I wanted to be displayed. Now I do not have any new row in the selection menu when I start the debug.

My config is:

```
  {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug J-Link App",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/BM96-CDK-33_8.elf",
            "serverpath": "JLinkGDBServerCLExe",
            "servertype": "jlink",
            "device": "SAM3A4C",
            "interface": "swd",
            "serialNumber": "", //if you have more than one J-Link probe add the serial number here 
            //"jlinkscript": "${workspaceRoot}/BSP/SEGGER/K66FN2M0_emPower/Setup/Kinetis_K66_Target.js",
            "runToMain": true,
            "svdFile": "${workspaceRoot}/.vscode/ATSAM3A4C.svd",
            "configFiles": [
              "interface/jlink.cfg",
                "target/at91sam3ax_4x.cfg"
            ],
            "serverArgs": [
                "-rtos",
                "GDBServer/RTOSPlugin_FreeRTOS"
            ],
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 86000000,
                "swoFrequency": 6000000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "label": "ITM0",
                        "port": 0,
                        "encoding": "ascii"
                    }
                ]
            }
        },
```

I checked the doc thinking about some change in the extension, but I did not found anything related to the swo config. Any hints?

thank you in advance

haneefdm, 2021-09-06:
Are you looking in the Terminal Tab? All SWO/RTT/Server outputs can now be found in the Terminal Window. That occurred about a month ago.

andyinno, 2021-09-06:
> Are you looking in the Terminal Tab? All SWO/RTT/Server outputs can now be found in the Terminal Window. That occurred about a month ago.

Nope! Just powered the computer after holidays and spent more or less 10 days without the SWO output. Hoping to find the reason in the issues or the wiki page. Today I decided it was the right time for writing here.

Thank you!
that's the solution.

closed 2021-09-06 by andyinno

haneefdm, 2021-09-06:
We do most of our change announcements in the CHANGELOG. But everyone is free to edit the Wiki as well.

andyinno, 2021-09-06:
Understood.
I checked wiki, issues and I did not found any mention about it.
I was just thinking that something was broken with my setup

Thank you for the reply.